greatscottgadgets/hackrf #1392: [Minor] HackRF One: Replace micro-USB with USB type C

raleighlittles, 2023-12-22:
### What feature would you like to see and why?

Can someone update the HackRF One to use a USB-c connector instead of a microUSB? The pinout and everything else can stay the same (no interface updates, etc), I just need it to have the USB-c connector.

Thanks for your consideration.

M0rn1ng574r, 2023-12-25:
> ### What feature would you like to see and why?
> 
> Can someone update the HackRF One to use a USB-c connector instead of a microUSB? The pinout and everything else can stay the same (no interface updates, etc), I just need it to have the USB-c connector.
> 
> Thanks for your consideration.

It's a very good idea, I thought the same thing, technically it's possible for charging, but wouldn't there be a problem for data transmission?

raleighlittles, 2023-12-25:
No, there shouldn't be a problem, USB-c is backwards compatible with the USB 2.0 interface.

[image]

You would just need to remove the micro USB connector, replace it with a USB-c one, and then reroute the 4 standard USB pins: VBUS, GND, D+ and D-. The other pins can just be set to low (I think, not 100% sure, but they _can_ be unused).

> It's a very good idea, I thought the same thing, technically it's possible for charging, but wouldn't there be a problem for data transmission?

CRD716, 2024-03-05:
Wonder if there would be any use for a 3.1 speed connection.

tdjastrzebski, 2024-04-14:
Changing connector to USB-C would be a very simple and very appreciated improvement.
Let's face it, USB-C gained wide acceptance everywhere and in 2024 it is really time to move forward.
Anyone who says that adapter could be uses clearly missies the point.
Note that, with more some improvements, this may allow for higher power supply voltage and more output power.

straithe, 2024-04-15:
USB-C is not a simple change to the board. As you've mentioned, there is a difference in power supply. That change alone would require a lot more engineering work than substituting one type of USB connector for another. If you want to understand the depth of changes required I recommend doing a deep dive into the different USB spec documents: https://usb.org/documents

raleighlittles, 2024-04-15:
@straithe I explained this in my original comment above.. the goal of this feature request was really just to replace the connector itself. You _can_ use the USB-c connector with the legacy USB 2.0 interface. Moving to the USB 3.0+ interface would, as you pointed out, require engineering work, but simply switching the connector type would just be a part replacement. All you would have to do is add a second set of V, G, D+, D- pin routes.

tdjastrzebski, 2024-04-15:
@straithe The change required to use full USB-C power delivery (PD) potential in sink mode, that is, ability to negotiate higher voltage and/or higher current is not a small change, likely additional chip like [TPS25730](https://www.ti.com/product/TPS25730) would be required to avoid dealing with CC lines directly.

However, the above is just an option. Simple socket change is much easier, would require just two 5.1k CC1/CC2 pull-down resistors and overshoot protection circuit (a TVS Schottky Diode and RC Snubber). To be super safe, half of the [LS24062RQ23](https://www.littelfuse.com/products/protection-ic/efuse/ls24062rq23) chip could be used. That is all what is needed.

ICantMakeThings, 2024-07-04:
https://github.com/ICantMakeThings/hackrf/blob/master/hardware/hackrf-one/hackrf-one.kicad_pcb
here, i also made a issue with USB-c
@raleighlittles

martinling, 2024-07-04:
We know that it would be possible to change the connector to USB-C, but there are some practical issues with doing so from a production and business point of view.

- We ship HackRF One with an injection molded enclosure, which has an opening sized for the smaller USB Micro-B connector. If we changed the board to use USB-C, we'd need to remake the mold tooling, which is expensive. We’d need to manufacture new enclosures, and discard all our existing stock of enclosures.

- There are many enclosures offered by other manufacturers, including all those designed for the various Portapack add-ons. Changing the connector type would break compatibility with those enclosures.

- Currently all revisions of HackRF One are equivalent and have the same SKU. This keeps things simple for stock keeping, returns, users integrating the device into other systems, etc. A connector swap would mean that newer and older units would no longer be interchangeable, which would add a lot of unnecessary complication to operations for GSG, our resellers, and end users.

It's hard to justify the upheaval of this change, when the USB-C connector doesn't offer any functional advantage for a USB 2.0 device. It's not going to get you faster speeds, or different power capabilities, or any other benefits except reversibility. We know some folks prefer all their things to be USB-C so that they can use their USB-C cables everywhere, but we also have a lot of users with a lot of Micro-B cables. We can't please everyone.

As such, we’re not planning to make this change. If you really want a HackRF with a USB-C connector, you're welcome to design and build your own. All our hardware is open source, so that everyone is free to make changes and remix our designs in ways that suit them better.

closed 2024-07-04 by martinling

Commander-Crash, 2024-07-21:
> It's a very good idea, I thought the same thing, technically it's possible for charging, but wouldn't there be a problem for data transmission?

I have tried this with a few usb-c breakout's it works for a moment but will get "SV#4" crash or total disconnection for no reason. I think its the way the USB clock is being generated IDK.

CoelacanthusHex, 2025-02-05:
> It's not going to get you faster speeds, or different power capabilities, or any other benefits except reversibility.

Actually, Type-C provides higher power capability although without Power Delivery, just puts two 5.1k Ohm Rd on CC. It's capable to 5V @ 3A while Type-A/Type-B/Micro-B only have 5V @ 1.5A power at maximum. 15W is double of 7.5W.

tdjastrzebski, 2025-02-05:
Usually max current @ 5V is even slightly higher: 3.15A.
It is possible because USB-C uses four parallel pins for Vcc and four pins for ground.