greatscottgadgets/hackrf #1395: HackRF r4-p2 won't stay unbricked

bgiblet, 2024-01-07:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

yes

### What outcome were you hoping for?

HacRF r4-p2 was purchased Jan 2017 from Sparkfun. Just got PortaPack H2+ from Hacker Warehouse, was trying to see something other than a black screen with no success. At some point this morning, the unit bricked and stopped showing up on USB. 
Have set aside the PortaPack and just want to get the bare HackRF going again. For this, am using an Ubuntu laptop. 

### What outcome actually happened?

Have managed to use DFU to load firmware distributions 2015.07.2, 2017-02.1, and 2023.01.1. The 2015 one didn't help. 2017 and 2023 both bring the USB port back online immediately, but USB drops out as soon as the unit is reset or powered down and the unit is not recognized. 

### What operating systems are you seeing the problem on?

Win7, Ubuntu Linux

### What is the output of ```hackrf_info```?

here's the load result:
willam@willam-Inspiron-3593:~/Downloads/hackrf-2017.02.1/firmware-bin$ dfu-util --device 1fc9:000c --download hackrf_one_usb.dfu --reset
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: DFU suffix CRC does not match
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1fc9:000c
Run-time device DFU version 0100
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 2048
Copying data from PC to DFU device
Download	[=========================] 100%        21992 bytes
Download done.
dfu-util: unable to read DFU status after completion
dfu-util: can't detach
Resetting USB to switch back to runtime mode

willam@willam-Inspiron-3593:~/Downloads/hackrf-2017.02.1/firmware-bin$ hackrf_info
hackrf_info version: unknown
libhackrf version: unknown (0.6)
Found HackRF
Index: 0
Serial number: 0000000000000000909864c82c356bcf
Board ID Number: 2 (HackRF One)
Firmware Version: 2017.02.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00554358

after reset, USB LED goes out and  it disappears

### Are you using any third-party software?

Was using Hackrf with SDR Console on a Win7 laptop, and that continued to work through several attempts at PowerPack installation. 

### Are you using any third-party hardware?

Portapack

bgiblet, 2024-01-09:
Really just wondering if anybody has suggestions as to what to try next. I'm wondering if the new firmware is not actually being written to non-volatile memory for some reason?

martinling, 2024-01-09:
Hi @bgiblet,

Using `dfu-util` will only load firmware into RAM. To write new firmware to flash you need to use `hackrf_spiflash`. See the [Updating Firmware](https://hackrf.readthedocs.io/en/latest/updating_firmware.html) section of the documentation for more info.

bgiblet, 2024-01-10:
Megathanks martinling! I seem to be back in business with the latest 2023 ver. It's communicating with both computers on USB.

Tried the Portapack again in hopes that modern firmware would work better with that. No joy so far.  That may end up as $200 worth of magic beans getting flogged on ebay. Maybe this HackRF r4 hardware is simply too old? I realize this isn't the place to deal with that though. Any wisdom you care to shed is appreciated.
Thanks again for helping a noob.
BG

bgiblet, 2024-01-13:
Martinling, no good deed goes unpunished. Not sure if I should start a new thread, but now when I start the SDR Console on my Windows machine, I get no signal, noise or otherwise. the unit is recognized on USB, and when I start the console, the RX led comes on, but no signals are coming through, dead quiet. I'm wondering if the same event that knocked it out before was a static zap that destroyed a key component. Any suggestions?

martinling, 2024-01-13:
Things to check if there's no signal:

- Is the antenna connected properly?
- Is the antenna connector actually RP-SMA instead of SMA?
- Have you turned up the gain settings?

bgiblet, 2024-01-13:
I have an antenna and was using it when receiving using SDR Console. Even without the antenna, I could see noise. Now there's nothing at all. I can try the gain settings, and see in the docs:  "A good default setting to start with is RF=0 (off), IF=16, baseband=16". 

I'm assuming we use the HackRF tools, but the docs don't go into any detail on how to check or set them. Is there a user guide I'm missing? 
Thanks, BG

straithe, 2024-04-15:
I suggest reading the documentation for SDR Console to learn how to change settings.