greatscottgadgets/hackrf #1400: FM modulation Block with Sawtooth Signal Source, I/Q Vs Time shows issues with Signal Generated.

smileyrekiere, 2024-01-11:
### What would you like to know?

O.k. have a simple vector generated off a sawtooth wave and using analog.frequency_modulator_fc,  where radians/sample = amplitude * sensitivity.  
Question: How to clean up IQ values for higher amplitude * sensitivity (radians/sample) for analog.frequency_modulator_fc?

At a high level modulating a signal (5 - 20 Mhz) on a carrier frequency in the 5 Ghz range with a pulse width from 50 - 100 micro seconds  the IQ values and modulated signal is not correct.

At lower frequencies (-2.5Mhz - 2.5 Mhz  to  -7.5 Mhz - 7.5 Mhz) the signal looks  as expected and the I/Q Vs Time looks close to what is expected. Between  -7.5 Mhz - 7.5 Mhz and -10 Mhz - 10 Mhz ,  the IQ values look incorrect and there is an anomaly on the edges of the modulated signal.

Lower  modulated frequency on carrier frequency

[image]
[image]

Higher modulated frequency on carrier frequency (I/Q values issue)

[image]
[image]

smileyrekiere, 2024-01-17:
Could the hackrf hardware affect the signal modulated signal?

Host at 2023 and hackrf_info at the following
hackrf_info version: git-13765055
libhackrf version: git-13765055 (0.8)
Found HackRF
Index: 0
Serial number: 000000000000000057b068dc22276763
Board ID Number: 2 (HackRF One)
Firmware Version: 2023.01.1 (API:1.07)
Part ID Number: 0xa000cb3c 0x00644754
**Hardware Revision: older than r6**                      
Hardware supported by installed firmware:
HackRF One

From the revisions see there is changes with the clock distribution
HackRF One r9[¶](https://hackrf.readthedocs.io/en/latest/list_of_hardware_revisions.html#hackrf-one-r9)
MAX2837 was replaced by MAX2839. Si5351C was replaced by Si5351A with additional clock distribution. A series diode was added to the antenna port power supply. Manufacturing year: 2023
HackRF One r9[¶](https://hackrf.readthedocs.io/en/latest/list_of_hardware_revisions.html#hackrf-one-r9)
MAX2837 was replaced by MAX2839. Si5351C was replaced by Si5351A with additional clock distribution. A series diode was added to the antenna port power supply. Manufacturing year: 2023

smileyrekiere, 2024-01-18:
What is the correct settings for the  osmosdr_sink_0.set_bandwidth(X,0)    some of the readings say to default to X to zero.  When doing so see the unusual I/Q values.  When I set the set_bandwidth( tx_sample_rate / 2, 0) see the signal is cleaned up a fair amount.  Other discussions mention .8 of the tx_sample_rate  (looking into other osmosdr_sink settings)

[image]

[image]

martinling, 2024-01-18:
What sample rate are you using?

smileyrekiere, 2024-01-19:
20 Mhz (wanting a chirp from -10 Mhz to 10 Mhz max to low of -2.5 Mhz to 2.5 Mhz)

That is the tx sample rate.

smileyrekiere, 2024-02-16:
Closing out.  The  I/Q issue is mitigated by having a precise equation  gnu radio uses radians/sample = amplitude * sensitivity.  Manually testing various values based on frequency.  The empirical values were then graphed and the resulting equation :  frequency_modulation * (pi /tx_sample_rate) this valure was then used as an input to the gnu  radio analog.frequency_modulator_fc  . 
[image]

smileyrekiere, 2024-02-16:
Closing as issue  resolved

closed 2024-02-16 by smileyrekiere