greatscottgadgets/hackrf #1407: How to set/check gain?

bgiblet, 2024-01-18:
### What would you like to know?

In my tech support post https://github.com/greatscottgadgets/hackrf/issues/1395, I was asked if I've tried setting the RF and IF gains. Haven't been able to find out how to check or set these. 
Is this possible to do this with a command line tool, or do we have to edit firmware source code, compile, and reflash?

Searching for this I saw Mossman's post from July 4, 2021 suggesting that this should be possible:
https://github.com/greatscottgadgets/hackrf/issues/906
But there are no replies, were these features ever implemented? I suspect all this is being done by volunteers, so I understand that maybe no one has taken the time. Just trying to figure out how these things can be done.

martinling, 2024-01-18:
These gains are set by whatever software you're using. In `hackrf_transfer` and `hackrf_sweep` they are command line parameters. In GNU Radio you set them on the sink or source block that you use. In graphical SDR tools they are normally presented as sliders.

bgiblet, 2024-01-20:
Thanks. It's pretty obvious to me now that something in the front end is hosed. I wasn't being too careful with static precautions while handling it, so it's probably my fault. I've used KICAD, so I <might> be able to view the schematics, poke around, and isolate the problem. I've done some SMD rework in my day job, but would put my success rate with fine pitch QFD's at about 25%. My instincts tell me to step out of the deep weeds, buy an off-the-shelf-radio, and write this off as a learning experience.