greatscottgadgets/hackrf #1408: DVT-T

iz5tep, 2024-01-25:
### What feature would you like to see and why?

Hi, I would like to know if it is possible to implement rtx in dvb-t.
DVB-S transmission is excellent, but DVB-T is missing at low ksr, 500/1000 ksr
Greetings
iz5tep

martinling, 2024-01-25:
This would be a feature request for the SDR software you're using. The HackRF project just provides the hardware and the raw I/Q sample interface.

iz5tep, 2024-01-25:
yes, that's exactly what I meant, I was referring to the software.
73

martinling, 2024-01-25:
What software are you using? You should open this issue there.

iz5tep, 2024-01-25:
software, SDRangel only dvb-s, both in rx and in tx with hackrf one
tnx

martinling, 2024-01-25:
Then you should open an issue on the SDRAngel repository: https://github.com/f4exb/sdrangel/issues

closed 2024-02-13 by martinling