greatscottgadgets/hackrf #1410: Second Signal with hackrf_transfer

Klammraute, 2024-02-03:
### Have you read the HackRF [troubleshooting documentation](https://hackrf.readthedocs.io/en/latest/troubleshooting.html)?

yes

### What outcome were you hoping for?

a normal Replay of a Signal

### What outcome actually happened?

When I try to send a .raw file, an additional signal appears to the right of the normal signal, even though the settings and the audio file are the same as with SDRAngel. It doesn't modulate the signal as desired. Did I forget something in the hackrf_transfer command? 
[image]

### What operating systems are you seeing the problem on?

Ubuntu Server

### What is the output of ```hackrf_info```?

hackrf_info version: unknown
libhackrf version: unknown (0.6)
Found HackRF
Index: 0
Serial number: 0000000000000000453c64c825655d8f
Board ID Number: 2 (HackRF One)
Firmware Version: 2017.02.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00644f5c

### Are you using any third-party software?

Hackrf Directly and SDRAngeln 

### Are you using any third-party hardware?

Normal Amateurradio Antenna

martinling, 2024-02-03:
You have the wrong sample format. The input to `hackrf_transfer` should be signed 8-bit samples. The SDRAngel documentation says it takes 16-bit samples.

Klammraute, 2024-02-03:
Hi, 

i have to correct me, the file i transmit with hackrf_transfer -t, i also recorded with hackrf_transfer -r, its a different file like in sdrangel

thank you

martinling, 2024-02-03:
Using `hackrf_transfer -r` will record all signals within the received bandwidth, including the DC spike in the middle. You would need to process the data further to extract the signal of interest.

closed 2024-05-22 by straithe