greatscottgadgets/hackrf #1413: Updated instructions to compile on Windows

raffaeler, 2024-02-11:
### What would you like us to add to or change about the HackRF One documentation?

This is not a request.
I just decided to post here the detailed and updated instructions to compile all the sources on a Windows machine. I added boring and obvious details which may anyway be useful for people not used to compile C++ sources on Windows.

I may create a pull-request, if Michael expresses the interest.

# How to compile `HackRF` on Windows

The first requirement (for this guide) is to install Visual Studio.

> No license is required because Microsoft provides the "build tools" version (command line compiler) or the "community edition" (Visual IDE and compiler) which are both free.

The second requirement is to install `CMake` from: https://cmake.org/download/. I tested my guide with the version `3.28.3`. When installing, I recommend to select the option that adds `CMake` into the user's `path` environment variable.

Note: It's totally possible to use `Visual Studio 2022` as well. You can also use the VS2022 IDE with the VS2019 C++ libraries. The Visual Studio installer gives you the possibility to install both.

After installing these two requirements, open a `developer command prompt` (a shortcut prepared by the Visual Studio installer). Be advised to open the command prompt only after completing the installation, as the environment variables are never updated in existing processes like the command prompt.

## Required libraries

Download and extract the following libraries inside the `c:\libs` folder.

You can use any folder, just carefully change the root folder in the batch files as well.

> Important: download the x64 version of all the dependencies

* `libusb`: https://github.com/libusb/libusb/releases (current version: 1.0.27)
* `pthreads`: https://sourceforge.net/projects/pthreads4w/files/ (pthreads-w32-2-9-1-release.zip)
  * There is a newer major version which I didn't test to avoid potential compatibility issues.
* `fftw`: https://www.fftw.org/install/windows.html ([fftw-3.3.5-dll64.zip](https://fftw.org/pub/fftw/fftw-3.3.5-dll64.zip) )

`fftw` does not provide a pre-built static library. This means that you will need to:

- generate the stub .lib (see next paragraph)
- always bring the `dll` file within the folder of the `exe` tools.

##Create the stub lib for `fftw`

From the `developer command prompt` for Visual Studio 2019. Be advised to use the same version of the command prompt of the compiler that you are going to use.

```
lib /def:libfftw3f-3.def /MACHINE:X64
```

These are not required, but just in case:

```
lib /def:libfftw3-3.def /MACHINE:X64
lib /def:libfftw3l-3.def /MACHINE:X64
```

## Clone the `HackRF` repository

The first step is to clone the sources from the GitHub repository.

```
git clone https://github.com/greatscottgadgets/hackrf
```

Then create the build folder inside the `hackrf` root folder:

```
cd hackrf\host
md build
cd build
```

The newly created build folder is used to host the `CMake` generated artifacts.

Now create the following two batch files inside the `build` folder.

## Batch 1: `mymake.cmd`

```
cmake .. -G "Visual Studio 16 2019" -A "x64" -DCMAKE_BUILD_TYPE=Release -DLIBUSB_INCLUDE_DIR=c:\libs\libusb-1.0.27\include -DLIBUSB_LIBRARIES=c:\libs\libusb-1.0.27\VS2019\MS64\static\libusb-1.0.lib -DTHREADS_PTHREADS_INCLUDE_DIR=c:\libs\pthreads-w32-2-9-1-release\Pre-built.2\include -DTHREADS_PTHREADS_WIN32_LIBRARY=c:\libs\pthreads-w32-2-9-1-release\Pre-built.2\lib\x64\pthreadVC2.lib -DFFTW_INCLUDES=c:\libs\fftw-3.3.5-dll64 -DFFTW_LIBRARIES=c:\libs\fftw-3.3.5-dll64\libfftw3f-3.lib
```

This batch file does uses Visual Studio 2019 to compile the solution. The options are:

| Option                                                       | Meaning                                                      |
| ------------------------------------------------------------ | ------------------------------------------------------------ |
| `-G "Visual Studio 16 2019"`                                 | Use VS2019                                                   |
| `-A "x64"`                                                   | Compile for x64<br />Previous `CMake` releases used `Win64` string |
| `-DCMAKE_BUILD_TYPE=Release`                                 | Compile a Release version                                    |
| `-DLIBUSB_INCLUDE_DIR=c:\libs\libusb-1.0.27\include`         | Include folder for `libusb`                                  |
| `-DLIBUSB_LIBRARIES=c:\libs\libusb-1.0.27\VS2019\MS64\static\libusb-1.0.lib` | Reference the `libusb` static library                        |
| `-DTHREADS_PTHREADS_INCLUDE_DIR=c:\libs\pthreads-w32-2-9-1-release\Pre-built.2\include` | Include folder for `pthreads`                                |
| `-DTHREADS_PTHREADS_WIN32_LIBRARY=c:\libs\pthreads-w32-2-9-1-release\Pre-built.2\lib\x64\pthreadVC2.lib` | Reference the `pthreads` static library                      |
| `-DFFTW_INCLUDES=c:\libs\fftw-3.3.5-dll64`                   | Include folder for `fftw`                                    |
| `-DFFTW_LIBRARIES=c:\libs\fftw-3.3.5-dll64\libfftw3f-3.lib`  | Reference the `fftw` stub library                            |

## Batch 2: `mycompile.cmd` 

```
msbuild hackrf.sln -p:Configuration=Release
copy .\libhackrf\src\Release\hackrf.dll .\hackrf-tools\src\Release
copy c:\libs\fftw-3.3.5-dll64\libfftw3f-3.dll .\hackrf-tools\src\Release
```

This batch does the following:

- Compiles all the source files:
  - `hackrf.lib`, `hackrf.dll`, `hackrf_static.lib`
    - The static library is not used by default, therefore we need to copy the `hackrf.dll ` to the tools target binary folder
  - All the exe tools 
- Copy the `hackrf.dll` into the tools target binary folder
- Copy the `libfftw3f-3.dll` into the tools target binary folder

## Compile

Run `mymake.cmd` to generate the `CMake` artifacts. Ensure you don't see any errors before proceding.

Finally run `mycompile.cmd` to compile the sources and copy the required libraries to the tools target folder.

# Generated binaries

All the binaries are generated in these folders:

* `hackrf\host\build\libhackrf\src\Release`

* `hackrf\host\build\hackrf-tools\src\Release`

straithe, 2024-02-12:
I suggest opening a PR to update the documentation!

raffaeler, 2024-02-12:
> I suggest opening a PR to update the documentation!

I can do it, but I saw the backlog and decided to wait for better times :)

martinling, 2024-02-13:
It should be possible to do this with a bit less work!

It's possible to install the dependencies in one command using [vcpkg](https://github.com/microsoft/vcpkg/):

`vcpkg install --triplet=x64-windows libusb fftw3 pthreads`

Unfortunately it still seems to be necessary to tell `cmake` where everything was installed, with something like:
```
cmake ..
    -DLIBUSB_INCLUDE_DIR=C:/vcpkg/installed/x64-windows/include/libusb-1.0 
    -DLIBUSB_LIBRARIES=C:/vcpkg/installed/x64-windows/lib/libusb-1.0.lib
    -DFFTW_INCLUDES=C:/vcpkg/installed/x64-windows/include
    -DFFTW_LIBRARIES=C:/vcpkg/installed/x64-windows/lib/fftw3f.lib
    -DTHREADS_PTHREADS_INCLUDE_DIR=C:/vcpkg/installed/x64-windows/include
    -DTHREADS_PTHREADS_WIN32_LIBRARY=C:/vcpkg/installed/x64-windows/lib/pthreadvc3.lib
```
(I'd like to get things tweaked so that cmake will discover these paths automatically if the relevant vcpkg variables are set.)

But then it should be possible to just `cmake --build . --config Release`

See also @dmaltsiniotis' work on #1391.

raffaeler, 2024-02-17:
@martinling My first wish is changing all the libraries to be statically linked so that you don't have to bring dlls with you.
That's not hard but I did not want to recompile fftw (apparently they do not provide static libs) or make changes to the cmake files.

FrankBolleri, 2024-11-14:
Hello everyone and excuse me for the question... but isn't there somewhere hackrf.dll and the hackrf tools already compiled for Windows in the latest version (2024.02.1?), ready to be used?
I would like to use them without having to create them myself with CMake etc...
I would have expected to find them in the Releases Assets here in GitHub, but I didn't find them.

Thanks in advance
Francesco

dmaltsiniotis, 2024-11-14:
Hi @FrankBolleri, as it turns out, yes there is. You can download per-compiled Windows binaries of the dll and tools from the nightly build process. The most recent run is located here: https://github.com/greatscottgadgets/hackrf/actions/runs/11778239935 and the link to download is at the bottom of the page:

[image]

Cheers,
-DM

FrankBolleri, 2024-11-14:
> Hi @FrankBolleri, as it turns out, yes there is. You can download pre-compiled Windows binaries of the DLL and tools from the nightly build process. The most recent run is located here: https://github.com/greatscottgadgets/hackrf/actions/runs/11778239935 and the link to download is at the bottom of the page:
> 
> ![image](https://private-user-images.githubusercontent.com/106233/386260935-1fa64d9c-88eb-4f37-9b2c-8dd95d35f6be.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzE1OTkzMTcsIm5iZiI6MTczMTU5OTAxNywicGF0aCI6Ii8xMDYyMzMvMzg2MjYwOTM1LTFmYTY0ZDljLTg4ZWItNGYzNy05YjJjLThkZDk1ZDM1ZjZiZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTE0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTExNFQxNTQzMzdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lYjlhMTllYjJjMWY2ZjZlNGM0MDFlODhjZTNjNTBiNTVkMjQxNThjNjZiYjM0YzE0MDIxZjY2MTllNzU3ZDhhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.rPe1fNagtszvNi5PsYPksomDnQl3ukNt2JL5koYFpDU)
> 
> Cheers, -DM

Thank you very much! Have a nice day!

martinling, 2025-10-04:
I've simplified the process for building on Windows a bit, and put a proposed update to the build instructions in PR #1595.

closed 2026-01-04 by mossmann