greatscottgadgets/hackrf #1414: hackrf_sweep hangs being run in cycle

egm108, 2024-02-12:
### What type of issue is this?

permanent - occurring repeatedly

### What issue are you facing?

I need sweep in cycle (to sweep different frequencies)
Both variants:
a) while true; do hackrf_sweep -1 -f 2400:2500; done > out 2>&1
b) c code when I do hackrf_start_rx_sweep and hackrf_stop_rx

lead to no sweeping.

for (a)
call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 2400 MHz to 2500 MHz
Stop with Ctrl-C
2024-02-12, 18:52:45.438607, 2400000000, 2405000000, 1000000.00, 20, -70.99, -61.34, -63.00, -63.37, -67.39
2024-02-12, 18:52:45.438607, 2410000000, 2415000000, 1000000.00, 20, -68.79, -61.94, -67.48, -56.33, -55.68
2024-02-12, 18:52:45.438607, 2405000000, 2410000000, 1000000.00, 20, -64.41, -63.51, -56.14, -63.15, -65.91
2024-02-12, 18:52:45.438607, 2415000000, 2420000000, 1000000.00, 20, -59.28, -61.79, -65.36, -72.95, -74.91
2024-02-12, 18:52:45.438607, 2420000000, 2425000000, 1000000.00, 20, -51.90, -49.43, -50.72, -59.27, -67.18
2024-02-12, 18:52:45.438607, 2430000000, 2435000000, 1000000.00, 20, -62.72, -64.52, -56.73, -50.79, -51.09
2024-02-12, 18:52:45.438607, 2425000000, 2430000000, 1000000.00, 20, -53.96, -55.98, -60.97, -59.57, -61.22
2024-02-12, 18:52:45.438607, 2435000000, 2440000000, 1000000.00, 20, -52.17, -53.34, -58.50, -60.74, -60.55
2024-02-12, 18:52:45.438607, 2440000000, 2445000000, 1000000.00, 20, -53.61, -56.22, -57.14, -56.08, -64.61
2024-02-12, 18:52:45.438607, 2450000000, 2455000000, 1000000.00, 20, -49.45, -55.27, -56.98, -55.46, -57.99
2024-02-12, 18:52:45.438607, 2445000000, 2450000000, 1000000.00, 20, -56.54, -73.85, -57.36, -55.30, -55.26
2024-02-12, 18:52:45.438607, 2455000000, 2460000000, 1000000.00, 20, -55.96, -61.23, -57.68, -55.75, -56.46
2024-02-12, 18:52:45.438607, 2460000000, 2465000000, 1000000.00, 20, -53.99, -56.38, -63.27, -60.91, -58.08
2024-02-12, 18:52:45.438607, 2470000000, 2475000000, 1000000.00, 20, -55.83, -55.09, -57.14, -57.72, -56.21
2024-02-12, 18:52:45.438607, 2465000000, 2470000000, 1000000.00, 20, -59.86, -59.66, -61.24, -56.41, -59.90
2024-02-12, 18:52:45.438607, 2475000000, 2480000000, 1000000.00, 20, -53.29, -58.89, -61.77, -58.44, -59.30
2024-02-12, 18:52:45.438607, 2480000000, 2485000000, 1000000.00, 20, -55.99, -54.57, -57.51, -68.44, -54.82
2024-02-12, 18:52:45.438607, 2490000000, 2495000000, 1000000.00, 20, -69.77, -53.48, -54.35, -64.10, -62.91
2024-02-12, 18:52:45.438607, 2485000000, 2490000000, 1000000.00, 20, -62.91, -59.65, -55.76, -57.72, -61.56
2024-02-12, 18:52:45.438607, 2495000000, 2500000000, 1000000.00, 20, -64.32, -59.21, -56.78, -53.10, -50.91

Exiting...
Total sweeps: 1 in 0.05023 seconds (19.91 sweeps/second)
hackrf_close() done
hackrf_exit() done
exit
call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 2400 MHz to 2500 MHz
Stop with Ctrl-C
0 total sweeps completed, 0.00 sweeps/second
0 total sweeps completed, 0.00 sweeps/second
0 total sweeps completed, 0.00 sweeps/second
.....

I debugged this a little bit. It happens that we cannot extract frequency in rx_callback, so the callback is called, but some trash is going inside buffer.
if (ubuf[0] == 0x7F && ubuf[1] == 0x7F) { I mean a condition is wrong, so we cannot extract frequency.

### What are the steps to reproduce this?

while true; do hackrf_sweep -1 -f 2400:2500; done > out 2>&1

My software see below.
You guys had some similar problems here...
But:
I see in my hackrf version (from github) I have all these fixes (checked not just date but lines from commits).... That's why I create a new one.

### Can you provide any logs? (output, errors, etc.)

egm@raspberrypi:~/ext_lib/hackrf/hackrf/host/build $ hackrf_info
hackrf_info version: git-4b31fd91*
libhackrf version: git-4b31fd91* (0.8)
Found HackRF
Index: 0
Serial number: 0000000000000000f75461dc2c9092c3
Board ID Number: 4 (HackRF One)
Firmware Version: 2023.01.1 (API:1.07)
Part ID Number: 0xa000cb3c 0x00614758
Hardware Revision: r9
Hardware does not appear to have been manufactured by Great Scott Gadgets.
Hardware supported by installed firmware:
    HackRF One
egm@raspberrypi:~/ext_lib/hackrf/hackrf/host/build $ ^C
egm@raspberrypi:~/ext_lib/hackrf/hackrf/host/build $ uname -a
Linux raspberrypi 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux
egm@raspberrypi:~/ext_lib/hackrf/hackrf/host/build $

egm108, 2024-02-12:
latest libusb, latest hackrf from github..

martinling, 2024-02-12:
Thanks for the report. As you note, this does look similar to some past issues, but you're running the latest code so there must still be something else going wrong.

Pokisan87, 2025-01-26:
I encountered the same problem, please tell me if there was a solution?