greatscottgadgets/hackrf #1417: Installed TCXO but still read from register 0 0x51

NazariyZ, 2024-02-21:
### What would you like to know?

Installed [TCXO ](https://www.ebay.com/itm/325998373482?var=515052966334&norover=1&mkevt=1&mkrid=711-167653-853678-5&mkcid=2&itemid=515052966334_325998373482&targetid=293946777986&device=c&mktype=pla&googleloc=9061018&poi=&campaignid=20143422176&mkgroupid=148565242265&rlsatarget=pla-293946777986&abcId=&merchantid=119648210&gad_source=1&gclid=CjwKCAiA29auBhBxEiwAnKcSqlpCJAaKRYmZiQFfxXOhWwhxJr7CQsMo4W3pbjOmfl3TXILfR1UC0xoCd2MQAvD_BwE)to my HackRf.
TCXO is working but when i'm reading register 0 i get 0x51 like there is no TCXO.
[image]

martinling, 2024-02-21:
You have a HackRF revision 9 board. The clock selection method is different to older revisions, so this register does not tell you anything about the clock status.

The better way to check the clock status is with `hackrf_clock -i`, which works for all revisions.

To get that command you will need a newer version of the host software, the PothosSDR builds are out of date.

You can find an unofficial Windows build of the latest release here:
https://github.com/dmaltsiniotis/hackrf/releases/tag/v2023.01.1

martinling, 2024-02-23:
Yes, based on the `hackrf_clock -i` result your TCXO is detected correctly.

The `hackrf_clock -a` command only lists the _output_ clocks of the Si5351 clock generator.

NazariyZ, 2024-02-23:
Thank you very much for your help.

closed 2024-02-23 by martinling