greatscottgadgets/hackrf #1422: Problem Updating to release 2024.02.1

AlexLandherr, 2024-02-29:
### What would you like to know?

I'm trying to update my HackRF One setup to release 2024.02.1 as written by first updating libhackrf and hackrf-tools like this:
```
alexl@PD70PNP:~/hackrf-2024.02.1/firmware-bin$ sudo apt install hackrf
[sudo] password for alexl: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
hackrf is already the newest version (2022.09.1-3).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```
I tried updating the firmware as written:
```
alexl@PD70PNP:~/hackrf-2024.02.1/firmware-bin$ hackrf_spiflash -w hackrf_one_usb.bin
File size 42248 bytes.
Checking target device compatibility
Unsupported Board IDCompatibility test failed.
```

And my board specs:
```
alexl@PD70PNP:~/hackrf-2024.02.1/firmware-bin$ hackrf_info
hackrf_info version: 2022.09.1
libhackrf version: 2022.09.1 (0.7)
Found HackRF
Index: 0
Serial number: 0000000000000000675c62dc304422cf
Board ID Number: 4 (HackRF One)
Firmware Version: 2023.01.1 (API:1.07)
Part ID Number: 0xa000cb3c 0x006a475f
Hardware Revision: r9
Hardware appears to have been manufactured by Great Scott Gadgets.
Hardware supported by installed firmware:
    HackRF One
```

Does this release not support my r9 board? Or is it something else?

mikemalpass, 2024-03-13:
I'm having exactly the same issue. Bought from nooelec (authorized reseller) on Amazon.

AlexLandherr, 2024-03-13:
I got mine from an authorized seller in the Netherlands as it's in the EU.

My r9 unit still works fine. I just want this fixed too like you do.

martinling, 2024-03-13:
The problem here is that your `hackrf_spiflash` program is from the 2022.09.1 release (the latest version packaged in Debian stable), and this version does not yet know about the r9 hardware.

You have two options, either of which will work:

- Install a newer version of the HackRF host software, either by building from source or using the package from Debian `testing`, which is now at 2023.01.1. The new `hackrf_spiflash` will recognise the r9 hardware.

- Override the compatibility check by passing the `-i` option to `hackrf_spiflash`. This is fine to do if you are sure the firmware image is compatible with HackRF One r9, i.e. it is the `hackrf_one_usb.bin` from the 2023.01.1 release or later.

AlexLandherr, 2024-03-13:
> The problem here is that your `hackrf_spiflash` program is from the 2022.09.1 release (the latest version packaged in Debian stable), and this version does not yet know about the r9 hardware.
>
> You [...]

So the override command would be this?:
 ```
alexl@PD70PNP:~/hackrf-2024.02.1/firmware-bin$ hackrf_spiflash -i -w hackrf_one_usb.bin
 ```

I intend to go with this since the firmware image downloaded is from the 2024.02.1 release.

martinling, 2024-03-13:
Yes that's right.

AlexLandherr, 2024-03-13:
It seems to have worked (after I unplugged and plugged it in again):
```
alexl@PD70PNP:~/hackrf-2024.02.1/firmware-bin$ hackrf_spiflash -i -w hackrf_one_usb.bin
File size 42248 bytes.
Erasing SPI flash.
Writing 42248 bytes at 0x000000.
alexl@PD70PNP:~/hackrf-2024.02.1/firmware-bin$ hackrf_info
hackrf_info version: 2022.09.1
libhackrf version: 2022.09.1 (0.7)
Found HackRF
Index: 0
Serial number: 0000000000000000675c62dc304422cf
Board ID Number: 4 (HackRF One)
Firmware Version: 2023.01.1 (API:1.07)
Part ID Number: 0xa000cb3c 0x006a475f
Hardware Revision: r9
Hardware appears to have been manufactured by Great Scott Gadgets.
Hardware supported by installed firmware:
    HackRF One
alexl@PD70PNP:~/hackrf-2024.02.1/firmware-bin$ hackrf_info
hackrf_info version: 2022.09.1
libhackrf version: 2022.09.1 (0.7)
Found HackRF
Index: 0
Serial number: 0000000000000000675c62dc304422cf
Board ID Number: 4 (HackRF One)
Firmware Version: 2024.02.1 (API:1.08)
Part ID Number: 0xa000cb3c 0x006a475f
Hardware Revision: r9
Hardware appears to have been manufactured by Great Scott Gadgets.
Hardware supported by installed firmware:
    HackRF One
alexl@PD70PNP:~/hackrf-2024.02.1/firmware-bin$
```

I can still use SDR++ to record signals or just listen to them through the speakers.

martinling, 2024-03-13:
Yes, if you just wanted the latest firmware and prefer to stick with the Debian stable packages for the host side, you can do that - everything will work, you'll just be missing a few software features that were added in the 2023/2024 releases (see the release notes for a list).

If you want the latest host software too, you can build and install that from the downloaded release folder. It will install into `/usr/local`, which should make it automatically take priority over the Debian version installed in `/usr`.

closed 2024-03-13 by AlexLandherr