greatscottgadgets/hackrf #1426: A request about waveform or code defined transceiver lib for libhackrf update

PatrickDowS, 2024-03-04:
### What feature would you like to see and why?

My name is Patrick, I would like to suggest for hackrf transceiver feature update.
(1) could use transmit function to transmit a pulse(or self-defined such as using mixer feature) and control RF switch receive the response from transmitted signal.  What I want to do is to make a wireless signal interrogator (waveform may be defined),  from the lib file I do not find function to define transmitted waveform or durations.
(2)the lib could be called by C#,  or java，C++...

HackRF work as wireless interrogator may have wide interesting applications.

PatrickDowS, 2024-03-05:
Hope the lib support windows.

dmaltsiniotis, 2024-03-05:
Hi @PatrickDowS,

What you're asking for can can probably all be accomplished with GNU Radio. The general principal of the HackRF One device and associated libraries is not to include use-case specific functionality, but rather to enable a broad range of uses by offering a generic API that you can hook into, or that other SDR software can utilize (such as the many SDR apps out there.)

To that end, there is in fact a `libhackrf` library published here that works on Windows, Linux, and macOS that you can directly import into c++ (and probably also into C# with the appropriate marshaling/wrappers). You could hypothetically code what you asking for above, however it's a long way from the modulation abstractions you mention to the quadrature I/Q data required by the SDR. That's why I mention GNU Radio might be a better option. Once you design your 'wireless signal interrogator' visually using GNU Radio Companion, it's compiled down to a python file that you can execute.