greatscottgadgets/hackrf #1428: Hackrf_sweep error at ranges above 160mhz

Klammraute, 2024-03-11:
### What type of issue is this?

permanent - occurring repeatedly

### What issue are you facing?

Hello, when I use hackrf_sweep and display the CSV file with plottsweep, all the PNGs above a scan range of 160MHz are intersected with white lines. Is there any way to avoid this?

### What are the steps to reproduce this?

hackrf_sweep -f 140:300 -l 32 -g 48 -w 50000 -r test1.csv 
plotsweep test1.csv example.png --power-min -60 --power-max -15.0 --colormap magma
works perfectly 
[image]

but not over 160 mhz 
hackrf_sweep -f 140:310 -l 32 -g 48 -w 50000 -r test2.csv 
plotsweep test2.csv example2.png --power-min -60 --power-max -15.0 --colormap magma
[image]

martinling, 2024-03-13:
This happens because the timestamp of the data can change midway through a sweep.

If you install the latest 2024.02.1 release, there is a new `-n` option to `hackrf_sweep`, which will use the same timestamp throughout one sweep, and that should eliminate this effect.

Klammraute, 2024-03-13:
thank you, solved :3

closed 2024-03-13 by Klammraute