greatscottgadgets/hackrf #1431: Rev9 Hardware freq setting

8uzeef, 2024-03-20:
### What would you like to know?

I am testing my new HackRF1 rev9, after updating to 2024.02.1., and using Gqrx-2.15.9-x86_64.AppImage
Use is ham radio 3.5 to 30 MHz
The Hardware freq. is adjustable only in steps of about 29Hz, both with internal xtal and Ext GPSDO 10MHz
For individual 1 Hz steps, it seems unknown what the frequency actually is , without using an external reference.
An example is setting 14095600 Hz for wspr.
Is this coarse setting only applicable to R9 using the Si5351a, or is it same for all versions?
Thanks

martinling, 2024-03-20:
The RF frequency stepping will be the same between r9 and other versions. There is not that much difference in the clocking arrangements.

In all revisions of HackRF One, both the MAX2837/2839 transceiver and the RFFC5072 mixer operate from a fixed reference frequency of 40MHz, which is obtained by ÷20 division of an 800MHz PLL within the Si5351A/C. That 800MHz frequency comes from multiplying either the 25MHz crystal reference by 32x, or the 10MHz external reference by 80x.

The Si5351A version only differs in how the internal/external clocks are switched, and how the sampling clocks are divided. The details can be seen in the block diagrams here:
https://hackrf.readthedocs.io/en/latest/hardware_components.html#block-diagrams

To achieve a target RF frequency set by the user, the HackRF firmware first selects an IF frequency, and then generates PLL settings for both the MAX283x transceiver's LO and the RFFC5072 mixer's LO. Because of the complexity of the selection and PLL programming there is not a fixed RF frequency step size, but 29Hz sounds quite normal.

And unfortunately yes, it is a shortcoming of the current software and firmware that there is not a way to retrieve the RF frequency that was actually achieved. For most purposes it is usually close enough, but I appreciate that there are situations where you will want to have the exact value that was tuned, and that's something we need to make available.

Note however that the RF tuning step is not a constraint on actual frequencies that can be received or transmitted, as any arbitrary frequency may be used in the digital baseband domain.

If you're able to measure frequencies accurately against an external reference, then my suggestion would be that you choose an RF operating frequency for which the HackRF achieves an exact match, ideally somewhere just below or above the band of interest, and then work at the necessary offsets from this in software.

8uzeef, 2024-03-20:
Martin, Thanks for the information.
I am contributing to HamSci ionospheric measurements, where FLDigi in FMT mode is used to measure transient deviations of 
100's of milliHz sometimes using WWV and CHU time standards.
 I have found that the HackRF1 when locked to GPSDO 10MHz (Leo Bodnar) is  stable on the time standards (unmeasurable drift using FLDIGI over 24+Hr periods).
With your info I will tabulate settings of the Hardware Freq. and the Rx offset so settings will be repeatable.

I have a Hermes14 also on the Leo Bodnar here doing wspr and FST4W which is tunable in 1Hz steps.
I will use that as a master for the HackRF and tabulate HackRF1 settings.

Thanks again for the timely and interesting information.

martinling, 2024-03-20:
Cool, glad we could help.

closed 2024-03-20 by martinling