greatscottgadgets/hackrf #1433: hackrf_info , SopySDRUtil, gr-soapy  different results for lisusb_claim_interface

smileyrekiere, 2024-03-22:
### What would you like to know?

gr-soapy question for hackrf

Using SoapySDR, SoapyHackrf, hackrf-2023.01.01,  gr-soapy from gnuradio  build from source. C++ and C (edited hackrf with prints).

gr-soapy: libusb_claim_interface result: -6  (resource busy)
SoapySDRUtil:  lisusb_claim_interface result: 0 
hackrf_info:  lisusb_claim_interface result: 0 

/** Resource busy */
    LIBUSB_ERROR_BUSY = -6,
    
    

**hackrf_info**  succeeded

hackrf_info version: git-6211fbb0*
libhackrf version: git-6211fbb0* (0.8)
Found HackRF
Index: 0
Serial number: 0000000000000000675c62dc302565cf
libusb_get_configuration result: 0 
set_hackrf_configuration result: 0 
**lisusb_claim_interface result: 0** 
Board ID Number: 4 (HackRF One)
Firmware Version: 2023.01.1 (API:1.07)
Part ID Number: 0xa000cb3c 0x00664f63
Hardware Revision: r9
Hardware appears to have been manufactured by Great Scott Gadgets.
Hardware supported by installed firmware:
    HackRF One

**SoapySDRUtil**  succeeded

user@user:~/git/gnuradio_git/fosdem/hackrf_sink/build$ SoapySDRUtil --probe
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device 
libusb_get_configuration result: 0 
set_hackrf_configuration result: 0 
**lisusb_claim_interface result: 0** 
[INFO] Opening HackRF One #0 675c62dc302565cf...
Desired_serial_number: 0000000000000000675c62dc302565cf usb_list_len: 10 
usb device: 0000000000000000675c62dc302565cf 
libusb_get_configuration result: 0 
set_hackrf_configuration result: 0 
lisusb_claim_interface result: 0 

**gr-soapy** failed
user@user:~/git/gnuradio_git/fosdem/hackrf_sink/build$ ./hackrf_sink
libusb_get_configuration result: 0 
set_hackrf_configuration result: 0 
lisusb_claim_interface result: 0 
[ERROR] hackrf_exit() failed -- one or more HackRFs still in use
[ERROR] SoapySDR::Device::enumerate(hackrf) stoi
Desired_serial_number: 0000000000000000675c62dc302565cf usb_list_len: 10 
usb device: 0000000000000000675c62dc302565cf 
libusb_get_configuration result: 0 
set_hackrf_configuration result: 0 
**lisusb_claim_interface result: -6** 
[INFO] Could not Open HackRF Device: 0000000000000000675c62dc302565cf  dev: (null)  ret: -1000
[ERROR] hackrf_exit() failed -- one or more HackRFs still in use
terminate reached from thread id: 7fdfd4e2cb80Got std::runtime_error
hackrf open failed

So why does lisusb_cliam_interface fail with gr-soapy  ?  All scripts run in user space.

Any insight would be helpful

martinling, 2024-03-22:
This error happens when the interface is already opened by another program. Did you have more than one program open at once that might have been trying to use the same device?

smileyrekiere, 2024-03-23:
Martinling - thankyou 

In investigating if there was another program running added print statements to the hackrf.cc and SoapyHackRF code.  Noticed between the working and non-working code that kwargs for the device in gr-soapy needed to be for the device  "device=hackrf"  , previously had it set to "hackrf" .   The gr-soapy (and SoapyHackRF) had discovered the device because had set  dev_args to "serial=0000000000000000675c62dc302565cf" so the hackrf could be discovered.  

So to claim the interface "device=hackrf" needed to be set.

smileyrekiere, 2024-03-23:
Closing:  for gr-soapy need to have kwargs use "key value pairs" in C++ as is done in python from what I can tell.

closed 2024-03-23 by smileyrekiere