greatscottgadgets/hackrf #1435: Hackrf frequency Limits  Tx/Rx

smileyrekiere, 2024-03-25:
### What would you like to know?

Saw posted discussion.
https://github.com/greatscottgadgets/hackrf/issues/1381

From the discussion in 1381, if the hackrf host code tells the hackrf firmware to be configured configured to transmit or receive greater than 6 Ghz,  the hackrf will set a gpio pin that would set the external 10 Ghz Transverter to be enabled/ chosen  which with the external mixer and 4 Ghz Local Ossilator will upconvert for Transmit and downconvert for receive  the received signal by 4 Ghz.

Note per @martinling  the hackrf firmware would need to be updated to allow frequencies greater then 7.25 Ghz . And something like and Opera cake would be needed to select the antenna to receive on. 

Questions 
1.  Am I understanding the discussion from 1381 correctly?  The hackrf could transmit and receiver at  6 Ghz with something similar to the 10 Ghz Tranverter discussed?
2. Also in the discussion it @martinling mentioned 7.25 Ghz,  can the hackrf transmit and receive at 7.25 Ghz ?
3. Would there need to be amplification between the hackrf and the external transverter?
4. Are there any recommended low cost 10 Ghz Tranverters for hackrf?

martinling, 2024-03-26:
> Am I understanding the discussion from 1381 correctly? The hackrf could transmit and receiver at 6 Ghz with something similar to the 10 Ghz Tranverter discussed?

In principle, any radio can be used with a transverter if the input/output frequency and power levels are compatible. 

Operating will not necessarily be straightforward though. Unless the radio has specific support for a transverter, operation will require manual calculation of frequency offsets and manual TX/RX control.  We don't currently provide any specific transverter support in HackRF. The user in #1381 was planning to automate their setup via their own modifications to the HackRF firmware and software.

> Also in the discussion it @martinling mentioned 7.25 Ghz, can the hackrf transmit and receive at 7.25 Ghz ?

The HackRF's rated frequency range is 1MHz to 6GHz and it should not be expected to perform outside this range, but if you override the software limits and tell the firmware to tune up to 7.25GHz, it will do so. You should expect little or no output power, and heavily attenuated reception if any, but if you are determined to get some limited operation above 6GHz it might work.

> Would there need to be amplification between the hackrf and the external transverter?

That depends on the transverter and what input power it requires.

> Are there any recommended low cost 10 Ghz Tranverters for hackrf?

For the receive side, there is a very low cost solution for 10GHz reception: a standard satellite TV LNB, which will downconvert 10GHz signals to a few hundred MHz where they can be received easily with the HackRF. A common use for this is to receive the downlink from the QO-100 geostationary amateur transponder at 10.489-10.499 GHz, which will appear at 739-749MHz after the 9750MHz LO conversion in a typical LNB.

For the transmit side, there is nothing specifically suitable that I'm aware of. For a low cost solution you would probably need to build your own.

There are a number of 10GHz transverter designs in the amateur radio community for operation in the 3cm band. Usually these are designed to work with with an existing amateur HF/VHF/UHF transceiver operating at 28MHz, 144MHz or 432MHz as the IF. With some additional gain and filtering at the input, the same designs could potentially be used with a HackRF.

A good starting point for reading would be [this article from Paul Wade W1GHZ](http://www.w1ghz.org/MBT/Simple_and_Cheap_Transverter_for_10_GHz.pdf).

smileyrekiere, 2024-03-26:
@martinling Thank-you for the detailed answer.

closed 2024-03-26 by smileyrekiere