greatscottgadgets/hackrf #1439: hackrf.h ### Example TX code utilizing the transfer and flush callbacks.  Needed to change line to void flush_callback(void * transfer, int success) {

smileyrekiere, 2024-04-05:
### What would you like to know?

**Compiling the Example TX code utilizing the transfer and flush callback that is in  in hackrf.h had compile error**
(2023 host code)

copied to another file rfgen_tx.c

Needed to change line
**void flush_callback(hackrf_transfer *transfer) {**
to 
**void flush_callback(void * transfer, int success) {**

To get the source to compile and run.

Otherwise would get the following error 
gcc -o rfgen_tx rfgen_tx.c -lm -lhackrf

```
rfgen_tx.c: In function ‘main’:
rfgen_tx.c:80:36: warning: passing argument 2 of ‘hackrf_enable_tx_flush’ from incompatible pointer type [-Wincompatible-pointer-types]
80 | hackrf_enable_tx_flush(device, flush_callback, NULL);
| ^~~~~~~~~~~~~~
| |
| void (*)(hackrf_transfer ) {aka void ()(struct )}
In file included from rfgen_tx.c:3:
/usr/local/include/libhackrf/hackrf.h:1150:21: note: expected ‘hackrf_flush_cb_fn’ {aka ‘void ()(void , int)’} but argument is of type ‘void ()(hackrf_transfer )’ {aka ‘void ()(struct *)’}
1150 | hackrf_flush_cb_fn callback,

```

Curious if others may have this issue or if I have something in my env that requires the line change?

Verified the signal transmitted and debug printed

hackrf_open
hackrf_open_setup
libusb_get_configuration result: 0
hackrf_open_setup: set_hackrf_configuration result: 0
libusb_claim_interface result: 0
hackrf_set_freq
hackrf_set_txvga_gain
hackrf_enable_tx_flush
hackrf_close

Also Thank-you for the Code Example provided by :
// Transmit a 440Hz triangle wave through FM (144.5MHz) using the libhackrf API
// Copyright (c) 2022 László Baráth "Uncle Dino" HA7DN https://github.com/Sasszem

martinling, 2024-04-11:
Thanks for spotting this. I've put a fix in #1442. 

Note that the first argument is not a pointer to a `hackrf_transfer`, it's whatever context pointer you passed as the third argument to `hackrf_enable_tx_flush`.

smileyrekiere, 2024-04-11:
Appreciate the Note,  trying to grasp the meaning.

In the example code (in hackrf.h)  the third argument is NULL  , 

    `hackrf_enable_tx_flush(device, flush_callback, NULL);`

from hackrf.h

### Flush callback
This callback is optional, and only applicable in TX mode. It get called when the last transfer is completed, and it's advisable to only stop streaming via this callback. This callback can be set using @ref hackrf_enable_tx_flush
 
 Is this why I need to pass full transfer buffer size (262144 samples) to get the hackrf to transmit data ?  For small number of samples need to pad out the buffer with zeros because the third argument for hackrf_enable_tx_flush has a NULL pointer as the third arguement?  
 
 
[image]

PR #1442 (2024-05-02): Correct prototype of flush callback in example code.
Fixes #1439.
closed 2024-05-02 by martinling