greatscottgadgets/hackrf #1445: Hackrf boot directly in usb mode if usb detected? or default to usb somehow

adminy, 2024-04-17:
### What would you like to know?

how can I ensure that my hackrf is always in usb mode when connected to the pi.
is there a serial command I can type to flip modes

or perhaps hackrf has usb detect to flip, or maybe there is a way to default to usb?

thanks

miek, 2024-04-17:
The standard HackRF firmware is always in USB mode. Are you asking about switching out of PortaPack mode? If so, this would be a question for whichever PortaPack firmware you're using and not something we're able to address in our firmware.

adminy, 2024-04-17:
so when HackRF is plugged into the Pi, I have to click, go all the way down to usb mode, click, confirm the yes, then it turns into usb mode. Otherwise by default its UI mode.

martinling, 2024-07-05:
As miek said, this is an issue to raise with the Mayhem folks. The Mayhem firmware includes a copy of our standard HackRF firmware as a submodule. When you select the USB mode it boots that firmware instead of the Portapack UI. If you want some other way to switch it over, that's something that would have to be added to their firmware. I gather recent versions include some USB functionality while in the Portapack UI mode, so it might be possible.

closed 2024-07-05 by martinling