greatscottgadgets/hackrf #1447: RF Scanning on HackRF One

greghains, 2024-04-24:
### What would you like to know?

Hi.
First up, I trust this is the right place to ask such a question. Apologies in advance if I have messed up.
I am trying to wade through tons of information when looking for RF scanning software for my HackRF One - much of it not always helpful - so I thought I'd come to the "source".
I prefer to use MacOS or Linux, but I can do Windows under a VM. Whilst I am no stranger to CLI tools, I would rather have a GUI before me that will allow me to scan newbie-style.  
What software do people use please?
Thankyou in advance.
Greg

Lpd738, 2024-04-25:
I would recommend SDR++ https://github.com/AlexandreRouma/SDRPlusPlus
Or hackrf-spectrum-analyzer https://github.com/pavsa/hackrf-spectrum-analyzer
Or my fav qSpectrumAnalyzer https://github.com/xmikos/qspectrumanalyzer

greghains, 2024-04-25:
Thankyou LPD738.

SDR++ I have seen before but my Mac wont run it due to being seen as suspicious software – phht. Lol
Might need to try it under Windows VM.

The other two I have not seen, so will give them a crack

Cheers,
Greg

Hard2Borrow, 2024-05-15:
Referring to the following 2 links will provide you with the mass of available 3rd party software:

[https://hackrf.readthedocs.io/en/latest/software_support.html](https://hackrf.readthedocs.io/en/latest/software_support.html)
Chapters 23/ 24(more so 24 in this case): [https://hackrf.readthedocs.io/_/downloads/en/latest/pdf/](https://hackrf.readthedocs.io/_/downloads/en/latest/pdf/)

Some commandline tools can be found [here](https://hackrf.readthedocs.io/en/latest/hackrf_tools.html) and on page 67 of the [pdf](https://hackrf.readthedocs.io/_/downloads/en/latest/pdf/)- which are found in the above links.

Some other 3rd party integration with the HackRF that I didn't see mentioned in those official locations relate to mobile apps- I've played around with some(Android ) - though not as robust/ feature rich as you'd get on your PC, still provide some nice results... 
...just search "hackrf" in either or app store and you will be presented with many options... 

Some items to consider with mobile(even on PC really) if the desired results aren't met

- make sure a good quality OTG cable is being used(both power & data)
- if a hub is being utilized and the HackRF isn't showing up, try a hub that is USB2... same goes with cable... 
- in a couple occasions I had to also instal the driver app which you'll know once connected... more than likely it will be a secondary app by the same original app creator.
- speaking of drivers, may as well add this, though for PC- if you are struggling with driver issues or your PC can't see your HackRF:

greghains, 2024-06-23:
Hi Hard2Borrow

My apologies for the delayed response.
Thankyou for this information. :)

Greg

straithe, 2024-06-24:
Closing issue as it looks to be resolved. Will re-open if original poster indicates otherwise.

closed 2024-06-24 by straithe