greatscottgadgets/hackrf #1452: 1.4.2 to 2.0.1 issues

discoriot, 2024-05-03:
### Have you read the HackRF [documentation](https://hackrf.readthedocs.io/en/latest/)?

yes

### What outcome were you hoping for?

Updated hackrf, i attempted the sd update ensuring it was fat32, unzipped all the new build files to the sd card and then turned off and inserted the sd card. once booted it did not show updated version in bottom left but shows all files under file manager.

### What outcome actually happened?

no change because file manager

### What operating systems are you seeing the problem on?

downloaded and updated sd card with ubuntu

### What is the output of ```hackrf_info```?

n/a

### Are you using any third-party software?

n/a

### Are you using any third-party hardware?

n/a

discoriot, 2024-05-03:
[image]
[image]

miek, 2024-05-03:
We don't maintain the mayhem firmware, you should ask on their repo instead: https://github.com/portapack-mayhem/mayhem-firmware

closed 2024-05-03 by miek