greatscottgadgets/hackrf #1457: Antenna working freq!

FadySamann, 2024-05-25:
### What would you like to know?

The working frequencies of the HackRF One range from 1MHz to 6GHz. 
However, neither ANT500 nor ANT700 works in the range of 5GHz to 6GHz! 
Please provide me with a solution to achieve these frequencies.

FadySamann, 2024-05-31:
> You can buy any antenna on internet, you can also use the antenna from an old router

Thanks for your comment. Are there any compatibility issues, such as the oms of the antenna or connectors differences? 
Moreover, can I connect the antenna through an extension wire, or do I have to connect it directly to the device?

Moji14, 2024-06-01:
Like many other RF devices HackRF One has an input impedance of 50 ohms. You can use any antenna that has the same impedance. Generally speaking almost all RF devices have 50 Ohms impendance. Avoid antennas made for TV systems since they have 75 Ohms impedance (it will work anyway but perform badly, specially on TX operations).
No antenna performs correctly on the entire bandwidth of the HackRf One, that means that you should use a specific antenna for the band that you're interested in.
Yes, you can use "extension wires" as you call them, any 50 Ohms cable will work although certain types of cables are more suitable depending on your configuration (like distance from hackrf to the antenna, frequency, and so on).
If you're just starting with Hackrf One, I think any cable will serve your needs at the moment. Once you get more experienced and your needs are more specific then you can get the appropriate cable for your application.

closed 2024-07-05 by martinling