greatscottgadgets/hackrf #1458: PPTC for LNA protection?

Botox200, 2024-05-26:
### What would you like to know?

I received a few 0.5A 72V PPTC components. Does it make sense to connect these to the antenna input as protection against overvoltage and high currents? 

I have now killed 6 Hackrf-Ones (one original and 5 clones) and i am only partially able to replace the smoked LNAs. They're damned  small for a standard soldering iron.

If this could be a safe solution i am gonna buy a sixth one.

Regards

Moji14, 2024-06-01:
Burning 6 Hackrf Onc it's such unfortunate to hear.
Are you able to detect why are you getting more than 72V at the input of your device?
Placing a PPTC device on your RX/TX line will not have a good effect on your device performance. Usually, a DC block could protect your device but if it's AC then you'll need to place a filter instead.
I'll recommend you to investigate what is the source of the overvoltage that's killing your devices. And then act upon it.

Botox200, 2024-06-02:
Thank you for the answer. A PPTC between Hackrf-One and Antenna in a metal housing as shielding would also cause losses in performance? What kind of losses?

If I had an explanation for the problem that the LNAs of many SDRs (4 NOOELEC SMART TEE or 6 HACKRF-ONE) burned, I wouldn't be posting here. I like to experiment with antennas in the frequency range from 1.5 GHz to 1.9 GHz. I always connect an LNA between the antennas and the Hackrf-One. These LNAs are powered by the bias tee of the Hackrf-One. The LNAs i used are specifically designed for SDRs with bias tee (like the NOOELEC LANA). But also they burned.

My last experiment was with a chinese LNA 2 to 2000 MHz, 5V input between Antenna and the Hackrf-One. I tried to receive DECT frequencies in the range of 1.8 GHz to 1.9 GHz. The Setup: a Bi Quad antenna - 5V LNA - Hackrf-One. The LNA is powered via a 150uH inductor from the output of the Hackrf-One to the VCC Plus of the LNA. Although the Bi Quad antenna is physically short-circuited, it should have an impedance at 1.9 GHz. The same applies to QFH antennas for the L-band.

I also tried to desolder and replace the LNAs from the Hackrf-One board, but my poor soldering skills destroyed the entire boards.

What I find remarkable is that the receiption with this antenna circuit works perfectly. However, three days later I want to repeat the experiment, switch on the Hackrf-One and - - - LNA and the LNA of the Hackrf-One are burned. No idea whether the inductance creates a voltage spike when i switched off. Unfortunately, my radio knowledge is not sufficient for an explanation.

Moji14, 2024-06-03:
A PPTC is kind of a dynamic impedance that varies with temperature. That means you will no longer mantain 50 Ohms impedance on your line. It will also increase parasitics inductance and capacitance, so your bandwidth will no longer be 6GHz.

From this image 
https://hackrf.readthedocs.io/en/latest/_images/block-diagram-r9.png
You can see that is specified 3.3V 50mA max, if you're exeeding those values that could be one of the causes of your devices burning up.
Also, the Hackrf have already a TVS and few bypass capacitors before reaching the LNA, that means that most likely you are feeding too much signal from your external LNA to the hackrf internal LNA. That might seem impossible to you since youre looking for signals in the microwave range, but if you're not filtering the signal from your antenna your external amplifiers might be picking up strong signals like radio stations, TV, GSM, LTE near by stations and that might be too much power to handle for the hackrf hardware.
In that case, if you really need a strong preamplifier you have to filter out all the undesired signals with a band pass filter for example.
I hope that helps you out.

Botox200, 2024-06-03:
Yes, i think i understood.

If I understand you correctly, the problem is not due to a signal that is too strong on the frequency I am receiving, but due to an insanely amplified secondary signal from another frequency.

Or in other words, the cause is the broadband amplifier, which receives an unwanted strong signal on a completely different frequency, amplifies it, and the Hackrf-One cannot control this signal because of its strength - it burns out.

Does that mean, conversely, that if I operate the Hackrf-One **without** an external broadband amplifier, i.e. with just one antenna, any damage is excluded?

Moji14, 2024-06-03:
It'll very rare if it happens. But it is possible. For example if your antenna is really close (matter of just couple of metres) to a broadcast station.
The thing is that with such strong signals, even if the hardware can resist it you'll notice it looking at the signal quality because you'll get a considerable amount of distortion, interference, product intermodulation. The spectrum of your signal will be a mess and practically unintelligible.

Aren't you capable of receiving the signal you wanted without amplification?
The thing is try to avoid any external amplification. If you really need amplification, ensure you get a good quality filter with strong attenuation on the rejected bands.

Botox200, 2024-06-03:
I'll first monitor the frequencies in my area so I can locate the strong signals.

I definitely need an LNA for satellites and DECT decoding.

Luckily I still have one working Hackrf-One left, so I will connect a passive 1.4 GHz High-Pass filter between antenna and LNA - for my range of interest. It seems to be an alternative, since I don't want to buy a Band-Pass filter for every single frequency.

BTW, Yesterday I had a PPTC between the Hackrf-One and the LNA to decode DECT frequencies at 1.9 GHz. Surprisingly, I didn't have any loss of reception, but I understood your thermical-dependent advice.

Thanks again

Moji14, 2024-06-03:
Very glad to hear that you solved the problem yourself.
The only thing remaining is that it will be great to save all those burnt units.
Try to get one of those hot air guns and a lot of flux to remove and replace those damaged parts.

Botox200, 2024-06-04:
I've already destroyed all the circuit boards with my soldering iron :(

Here is the measurement from yesterday made with an SDR and the telescopic antenna from the Hackrf-One. In your block diagram there was a maximum signal strength of -5dbm for the Hackrf-One board. When I look at the frequencies in the broadcast FM range at about 100 MHz , I'm already at this value - and this with the simple antenna.
[image]

Moji14, 2024-06-04:
That finally ties up all my suspicions, and we both can say that your broadband LNA was killing the SDRs. The PPTC you placed acted as an attenuator and that solved the problem. But a propper set up will be using the correct combination of filters, attenuators and amplifiers.

closed 2024-06-04 by Botox200