greatscottgadgets/hackrf #1460: Why hackrf_sweep use INTERLEAVED mode and discards data from a center of filter?

kubrack, 2024-06-12:
### What would you like to know?

Hi, 

As far as I can see from `host/hackrf-tools/src/hackrf_sweep.c` code, 
It takes samples with a center frequency at +7.5 MHz (at 20 Msps, under 15 MHz baseband filter) to produce a spectrum of 
(center - 7.5 MHz) .. (center - 2.5 MHz) = 5 Mhz
and 
(center + 2.5 MHz) .. (center + 7.5 MHz) = 5 Mhz
then repeats it with a +5 MHz shift, which covers all 20 MHz of the spectrum during one step (which is interleaved into two sub-steps i.e. frequency block transfers).

So at each sub-step we take data from the left and right 5 MHz edges of 15 MHz of baseband filter and the 5 MHz of center is discarded.

But according to https://hackrf.readthedocs.io/en/latest/sampling_rate.html
we should use a 5 or 5.5 MHz baseband filter (4:1 at 20 Mhz sampling rate) to take the center 5 MHz of spectrum.  

I saw comment https://github.com/greatscottgadgets/hackrf/issues/349#issuecomment-280379815 where was explained that it was implemented to remove the DC spike in the middle (thanks to @dominicgs ).

But in such case a significant adjacent spectrum interference isn't removed, is it? 

If we would use a 15 MHz baseband filter in LINEAR mode to take the center 5 MHz instead of two 5 MHz edges in INTERLEAVED mode, as described in https://hackrf.readthedocs.io/en/latest/sampling_rate.html. 
Was this way not used just due to performance impact?
Or would it also lead to the DC spike issue?
If so, is the way to get rid of both of the issues, even in the slower way?

Thanks in advance!

martinling, 2024-06-12:
If you used a baseband filter bandwidth of just over 5MHz, and operated in the linear tuning mode, capturing 5MHz of spectrum around the centre frequency at each 5MHz tuning step, then you would have three problems:

1. There would be a DC spike in the middle of each 5MHz block of spectrum captured (as well as various LF to MF noise nearby, which can couple into the analog baseband signals within the device).

2. There would be attenuation towards the edges of each 5MHz block, because the filter rolloff is not a hard edge; it starts to attenuate within the nominal passband.

3. Sweeping would be only half as fast as it is now (because you're capturing 5MHz of spectrum at each step, rather than 10MHz at each step).

The interleaving scheme avoids issues 1 and 2, by discarding both the middle and the edges of the frequency range that is captured at each step.

You could apply the same interleaving scheme within a 5MHz bandwidth, but then issue 3 would be even worse, because you'd only be capturing 2.5MHz of bandwidth at each step: sweeping would be four times slower.

Using the interleaving scheme with a 20MHz bandwidth allows us to capture 10MHz of clean spectrum, in two 5MHz chunks, at each tuning step.

It's true in principle that a slower sweep with a narrower bandwidth at each step could lower the noise floor, but I don't know if you'd really see much benefit in practice.

kubrack, 2024-06-12:
Hi @martinling, thank you for so quick and clear answer.

Could you please clarify the point `2. There would be attenuation ...` since it's also true for actual hackrf_sweep which uses 5 MHz at both edges of 15 MHz BPF in interleaved mode.

Am I right that it isn't so significant as in the case of using center 5 MHz of 5 MHz BPF since 15 MHz BPF has a higher Q-factor than 5 MHz BPF?

I mean that according to max2837/max2839 RX BB FREQUENCY RESPONSE function
15 MHz RF BPF rejection at +/-7.5 MHz is ~ -1.5dB
while 
5 MHz RF BPF rejection at +/-2.5 MHz is ~ -2.5dB

martinling, 2024-06-12:
You've looked at this more closely than I have!

I hadn't realised we were only setting it to 15MHz. That will indeed give a bit of attenuation at the edges of each block.

Unfortunately though the next setting up is 20MHz, and that would have the issue of aliasing signals from just outside the 20MHz sampling bandwidth. But since we're throwing away both the centre and the edges, where those aliases would appear, that could be acceptable.

The other thing we could do is take advantage of the LPF fine-tuning register, and use 15MHz + 10% to get a 16.5MHz bandwidth.

kubrack, 2024-06-12:
Good idea.

AFAICS the register is already defined for max2839 https://github.com/greatscottgadgets/hackrf/blob/d4ed8474e532ccdbf605ee71c0f1fa0557da19e1/firmware/common/max2839_regs.def#L55

and for max2837 even with values:
https://github.com/greatscottgadgets/hackrf/blob/d4ed8474e532ccdbf605ee71c0f1fa0557da19e1/firmware/common/max2837_regs.def#L94-L97

Do you think it worth to make PR for it? If so I would try to make it later. 

Anyway, thank you for helping me understand the basic things above!

martinling, 2024-06-12:
Well, the first step would be to add firmware support for using the fine tuning registers to get finer-grain control of the LPF bandwidth. That seems like an improvement worth making on its own, although we'd want to do some tests to confirm that the effect is as expected.

In terms of code, that would involve extending the `max2837_ft` table to include the additional possible bandwidth values, with the `MAX2837_dF` settings for them as well as the `MAX2837_FT` settings, then writing the additional field in `max2837_set_lpf_bandwidth`. And similarly for the MAX2839. But fortunately there is no host-side or USB API change required, since the host just sends a target bandwidth in Hz.

Then with that change in place, it would be possible to test whether sweep mode is improved by using a 16.5MHz rather than 15MHz bandwidth. I suspect you might be hard pushed to detect the difference, but it's worth a try if you're interested.

closed 2024-07-25 by straithe