greatscottgadgets/hackrf #1461: Can't enable BiasTee in GNU radio for HackRf-One and RTL-SDR.

MuhammadQaisarAli, 2024-06-13:
### What would you like to know?

I am trying to transmit and receive a sine wave with hackrf as tx and rtl-sdr as rx. Hackrf is the latest hardware, The Latest GNU radio on Windows 10 was installed., in May 2024, while rtl-sdr is the v3 model.  While using a passive antenna, I can tx and rx fairly easily.

If I connect the active antenna, like a GPS antenna [For example this one](https://www.aliexpress.com/i/32862303898.html) I don't get the results. It appears like the signal is not transmitting via the antenna and is leaking at sma connectors of tx and RX.

probably it might be because of not providing power to LNAs in antennas, I searched a lot but wasn't able to enable/disable bias-tee in gnu radio but had no success. Although I was able to enable it for rtl-sdr in SDRAngel software.

I passed ` "SignalSource.osmosdr_args=rtl,bias=1" `, `"SignalSource.osmosdr_args=hackrf,bias=1" ` and `"SignalSource.osmosdr_args=hackrf,bias_tx=1" ` as device arguments to sink and source blocks. But it didn't work at all. 

If anyone knows the fix, please advise.
`

Moji14, 2024-06-13:
Where you able to receive signal using SDRangel?
The GPS antenna you referenced has a non precise electrical specifications datasheet. But it seems to be centered around 1570 MHz. So your receiving range will be more or less from say 1560 to 1580 MHz, roughly speaking. To be more accurate this antenna sole purpose is to receive signals from 1575.42(GPS) and 1561.098 (BeiDou). You must place it outdoors and have clear sky above it to receive from GPS or BeiDou.
The electrical specs don't specify current consumption either. So if your GPS antenna needs more that 50mA it won't work either. Most of those antennas don't require that much current but you should double check.

martinling, 2024-06-13:
You can't transmit into a typical active GPS antenna. They are receive-only devices, with an LNA connected directly between the patch antenna and the coax port. There's no TX path. If you try to transmit into it, your signal will just be running into the output port of the LNA.

MuhammadQaisarAli, 2024-06-28:
> You can't transmit into a typical active GPS antenna. They are receive-only devices, with an LNA connected directly between the patch antenna and the coax port. There's no TX path. If you try to transmit into it, your signal will just be running into the output port of the LNA.

Sorry for incomplete information from my side. You are totally correct that these are Rx only, and I was using pasive antenna for Tx.

Problem is that I can't enable bias-T and also with passive antenna the signal seems leaked.

MuhammadQaisarAli, 2024-06-28:
> Where you able to receive signal using SDRangel? The GPS antenna you referenced has a non precise electrical specifications datasheet. But it seems to be centered around 1570 MHz. So your receiving [...]

Actually for the time being i am not interested in real GPS, rather I was transmitting a narrowband (unlike GPS wide band below noise floor) signal at L1. I was able to receive it with those antennas with the feature of enabling and disabling Bias-T.  But i was unable to enable/disable Bias-T in GNU-Radus for rtl-sdr.

martinling, 2024-06-28:
Note that in the latest [2024.02.1 release] of the HackRF firmware and host tools, there is a new `hackrf_biast` utility that can be used to pre-configure bias tee behaviour, for use with software that doesn't support controlling it directly.

closed 2024-07-25 by martinling