greatscottgadgets/hackrf #1462: "hackrf_close() failed: Input/Output Error (-1000)"

Ross966, 2024-06-14:
### Have you read the HackRF [documentation](https://hackrf.readthedocs.io/en/latest/)?

yes

### What outcome were you hoping for?

Successful operation of hackrf_transfer and/or hackrf_sweep tool(s).

### What outcome actually happened?

Both hackrf_transfer and hackrf_sweep failed with message "hackrf_close() failed: Input/Output Error (-1000)".

Logs:

C:\Program Files\PothosSDR\bin>hackrf_transfer -r c:\dump\dump.rtf
call hackrf_set_sample_rate(10000000 Hz/10.000 MHz)
call hackrf_set_hw_sync_mode(0)
call hackrf_set_freq(900000000 Hz/900.000 MHz)
Stop with Ctrl-C
 0.0 MiB / 1.004 sec =  0.0 MiB/second, amplitude -inf dBfs

Couldn't transfer any bytes for one second.

Exiting... hackrf_is_streaming() result: streaming stopped (-1003)
Total time: 1.00440 s
hackrf_stop_rx() failed: Input/Output Error (-1000)
hackrf_close() failed: Input/Output Error (-1000)
hackrf_exit() done
fclose() done
exit

C:\Program Files\PothosSDR\bin>hackrf_sweep -f 2400:2490
call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 2400 MHz to 2500 MHz
Stop with Ctrl-C

Exiting... hackrf_is_streaming() result: streaming stopped (-1003)
Total sweeps: 0 in 0.05254 seconds (0.00 sweeps/second)
hackrf_close() failed: Input/Output Error (-1000)
hackrf_exit() done
exit

Additional info: the hackrf_clock, hackrf_spiflash and hackrf_debug tools work as expected

### What operating systems are you seeing the problem on?

Windows 11

### What is the output of ```hackrf_info```?

hackrf_info version: git-7047252
libhackrf version: git-7047252 (0.6)
Found HackRF
Index: 0
Serial number: 000000000000000057b068dc21829763
Board ID Number: 2 (HackRF One)
Firmware Version: 2024.02.1 (API:1.08)
Part ID Number: 0xa000cb3c 0x005f435c

### Are you using any third-party software?

N/A

### Are you using any third-party hardware?

N/A

martinling, 2024-06-14:
From the output, it seems like the device fell off the USB bus right as it went into RX mode. Bad USB cable perhaps?

Ross966, 2024-06-14:
Thats fixed it! I tried two differnt USB cables before but of the same (and presumably low) spec. I tired a new USB cable and it works. Easy fix. Thanks you!

martinling, 2024-06-14:
Great! Glad to have an easy one!

closed 2024-06-14 by martinling