greatscottgadgets/hackrf #1463: Replacement of TPS62410 Buck converter

anxzik, 2024-06-22:
### Have you read the HackRF [documentation](https://hackrf.readthedocs.io/en/latest/)?

yes

### What outcome were you hoping for?

I have a unit from 2014, when plugged in, it is recognized after a few moments its starts flickering LEDs and looses connectivity,  the chip in the corner gets extremely hot. using Kicad iv found the chip to be the TI Buck converter, yet the one on the unit is marked CAT 081 A5JK .. I just wanted to confirm that https://www.ti.com/product/TPS62410 is the correct chip  or if there had been a revision what part should i order?

### What outcome actually happened?

nothing yet seeking further guidance in repair of PCB

### What operating systems are you seeing the problem on?

Linux, Win 10 - 11

### What is the output of ```hackrf_info```?

n/a

### Are you using any third-party software?

No

### Are you using any third-party hardware?

Hack RF was Originally paired with Portapack

straithe, 2024-06-22:
Can you please add a picture of your HackRF to this issue?

martinling, 2024-07-05:
TPS62410 is the correct replacement for U21, regardless of board revision.

However, you have other issues: one of the capacitors next to U21 is visibly cracked:

[image]

I suggest you start by checking that whole area for other damaged components.

closed 2024-10-17 by straithe