greatscottgadgets/hackrf #1465: Add USB-c (pcb already modified)

ICantMakeThings, 2024-07-04:
### What feature would you like to see and why?

https://github.com/ICantMakeThings/hackrf/blob/master/hardware/hackrf-one/hackrf-one.kicad_pcb

Doubt it'll be accepted, but worth a shot. iv only updated the PCB, not the BOM with the USB-c and the 2 resistors for the cc pins, but the footprints are there for PD support.

martinling, 2024-07-04:
We're not planning to change to USB-C - please see the response at https://github.com/greatscottgadgets/hackrf/issues/1392#issuecomment-2209345108.

closed 2024-07-04 by martinling

Commander-Crash, 2024-07-21:
> We're not planning to change to USB-C - please see the response at [#1392 (comment)](https://github.com/greatscottgadgets/hackrf/issues/1392#issuecomment-2209345108).

Apple said the same lol