greatscottgadgets/hackrf #1467: what caused hackrf_transfer errors?

dudelaksdfjlkaz0102, 2024-07-16:
### Have you read the HackRF [documentation](https://hackrf.readthedocs.io/en/latest/)?

no

### What outcome were you hoping for?

hackrf_transfer -f 96900000 -t ./against_wind.C16 -a 1
hackrf_transfer: symbol lookup error: hackrf_transfer: undefined symbol: hackrf_enable_tx_flush

hackrf_transfer -s 250000 -f 98900000 -t against_wind.C16
argument error: sample rate hz should be greater than or equal to 2000000 Hz/2.00 MHz

### What outcome actually happened?

I tried the following commands and got the following errors. I'm not certain of the cause.
hackrf_transfer -f 96900000 -t ./against_wind.C16 -a 1
hackrf_transfer: symbol lookup error: hackrf_transfer: undefined symbol: hackrf_enable_tx_flush

hackrf_transfer -s 250000 -f 98900000 -t against_wind.C16
argument error: sample rate hz should be greater than or equal to 2000000 Hz/2.00 MHz

### What operating systems are you seeing the problem on?

Raspberry Pi OS (Legacy) Lite
    Release date: July 4th 2024
    System: 64-bit
    Kernel version: 6.1
    Debian version: 11 (bullseye)

### What is the output of ```hackrf_info```?

libhackrf version: unknown 0.5
Found HackRF
Index: 0
Serial Number: censored
Board ID Number: 2 (HackRF One)
Firmware Version: local-de927bd (API: 1.03)
Part ID Number: 0xa000cb3c 0x00584763    

### Are you using any third-party software?

NO

### Are you using any third-party hardware?

NO

martinling, 2024-07-18:
The first problem you have is:
`symbol lookup error: hackrf_transfer: undefined symbol: hackrf_enable_tx_flush`

You have a `hackrf_transfer` program that was built from a recent release, but the `libhackrf` library on your system is an old version, so `hackrf_transfer` is trying to use functions that aren't available.

You need to install the corresponding `libhackrf`, from the same release that you got `hackrf_transfer`.

The second problem you have is:
`argument error: sample rate hz should be greater than or equal to 2000000 Hz/2.00 MHz`

It's not a good idea to use sample rates less than 2MHz with HackRF, because the narrowest baseband filter available on the hardware is 1.75MHz wide.

You have IQ data with a 250kHz sampling rate that you'd like to transmit. The best thing to do would be to go back to how you generated that data, and regenerate it with a 2MHz sample rate or higher.

If that's not possible for whatever reason, then you can use an interpolating resampler in GNU Radio or some other tool, to interpolate the data to a 2MHz sample rate or higher.

straithe, 2024-08-15:
I'm closing this as there hasn't been a response in over 30 days. Please re-open this issue or open a new one if you still need assistance.

closed 2024-08-15 by straithe

dudelaksdfjlkaz0102, 2024-10-07:
I have an IQ file recorded at 250kHz sampling rate created with HackRF Mayhem v1.4 firmware. I'm unable to record the signal again. I need to convert the IQ to play at 2MHz sampling rate allowing it to replay with latest HackRF Tool & Mayhem firmware. Can you tell me the names of software that will convert and where I can find them?