greatscottgadgets/hackrf #1476: Only I or Q channel carrier wave transmit?

luoGanda, 2024-09-02:
### What would you like to know?

can this be done?
maybe am i missing something that already works in this direction?

Don't know much about hardware underneath,
but as i see it - one carrier wave channel is passed as it is and the other 90deg outOfPhase(and this is how it is read also).

One can modulate carrier wave with I and Q values, but what about carrier without modulation?
Eg putting 0 for I channel and any value in Q - can this vars be accessed somehow?
Is there a way to do this? if not - maybe via adding simple firmware code?

I wonder if portapack Utility->Signal generator does this?

Not sure if this would be an easy task to code, but it would be an handy option.

martinling, 2024-10-17:
To produce a carrier wave at the centre frequency, you can just output constant values for I and Q. The amplitude of the carrier wave will be proportional to the magnitude of the combined IQ vector, and its phase will be related to the angle of the IQ vector.

This what is done when you use `hackrf_transfer` with the `-c` option. It outputs the `-c` parameter as a constant I value.

luoGanda, 2024-10-19:
Yes, this would be it.
I almost don't know how have i missed it.
I checked hackr_transfer.c and -c option does what you say - I ch only.
I have checked docs to see which channel is delayed by 90deg - didn't find it, but it's probably Q,
so writing in I(like c option) is probably what i want.

In this way hrf kinda works as a funcGen(known as source signal), is this correct?

By the way - if someone is interested:

- [here](https://sites.google.com/view/ws-tools) at 1st link is a freeVer tool on linux to map signals on google maps.
This is for rtlsdr - but for hrf it would be probably easily modified.
A few years outdated but probably still works.
I have it also for windows, altough didn't check it if it works correctly.
There is a simple gui that captures signals in apropriate format(and also has some other small features)
- on the same site, i will probably put a software based hackrf sweep transmit tool in the future - since firmware code for hardware is not yet implemented(there is an thread/"issue" for that somewhere).
Not so fast as hardware would be, but fast enough for eg simple tests.

martinling, 2024-10-20:
> If one would put 0 to both transfer->buffer I and Q, there would still be carrier wave doing it's thing, isn't that correct?

No.

The output of an IQ modulator is the sum of one carrier that is multiplied by I, plus another carrier 90 degrees out of phase that is multiplied by Q, like this:

[image]

Or in symbolic terms, something like `I*sin(w*t) + Q*cos(w*t)`.

So if I and Q are both zero then the output is also zero.

Moji14, 2024-10-20:
> I have checked hackrf_transfer.c and -c option, but...
> are not here I and Q values just to modulate underneath carrier wave?
> If one would put 0 to both transfer->buffer I and Q, there would still be carrier wave doing it's thing, isn't that correct?
> Because, if one reads samples from file - they are "imprinted" on carrier wave, no matter what values are there.
> Aren't this I and Q amplitude values simply just for making - eg sine wave - which is later 'zipped'/modulated with carrier wave?
> 
> I have checked docs to see which channel is delayed by 90deg - didn't find it, but it's probably Q,
> so writing in I(like c option) - what you proposed - could be what i want.
> 
> In this way(if it would work like i asked for)
> hrf would kinda work as a funcGen(known as source signal), is this correct?
> 
> 
> By the way - if someone is interested:
> 
> - [here](https://sites.google.com/view/ws-tools) at 1st link is a freeVer tool on linux to map signals on google maps.
> This is for rtlsdr - but for hrf it would be probably easily modified.
> A few years outdated but probably still works.
> I have it also for windows, altough didn't check it if it works correctly.
> There is a simple gui that captures signals in apropriate format(and also has some other small features)
> - on the same site, i will probably put a software based hackrf sweep transmit tool in the future - since firmware code for hardware is not yet implemented(there is an thread/"issue" for that somewhere).
> Not so fast as hardware would be, but fast enough for eg simple tests.

Theoretically It's possible. But in practice it's more complicated. Just keep in mind that I and Q multiplies the LO, therefore multiplying by 0 you'll get 0. 
What you can do is for example set to 1 I and to 0 Q this way you'll get only the LO signal which will be your 'desired' carrier.
Check the block diagram of hackrf. This device uses double conversion. The first frequency conversion happens at U17 (MAX2839 or MAX2937) and the second one at U4 (RFFC5072) and depending on your desired signal you'll be using upconversion downconversion or bypassing U4.

luoGanda, 2024-10-21:
ok,
from both answers - if i understand correctly...

carrier wave is already there and writing any constant value in I(and 0 in Q or other way arround) would simply 'scale' carrier,
meaning that suggested -c option is doing the same thing?

@Moji14: that up/down-conversion is too much for me right now,
would this mean that theoretically one could go beyond 7.2GHz without much modification?

martinling, 2024-10-21:
> meaning that suggested -c option is doing the same thing?

Yes, as I said in my first answer - the `-c` option to `hackrf_transfer` just outputs a constant I value, which scales the carrier.

> would this mean that theoretically one could go beyond 7.2GHz without much modification?

You would need major modifications to the whole RF path to get useful performance beyond 7GHz.

HackRF One is designed and tested for use up to 6GHz.

The `hackrf_transfer` program allows the user to force the operating frequency as high as 7.2GHz because in practice there is still measurable output power when it is driven to that point, and that functionality may sometimes be useful even though it is well outside the design range.

In theory, you can get an 8.4GHz carrier by explicitly setting the IF to 3GHz, the mixer LO to 5.4GHz, and the image rejection to the high pass mode:

`hackrf_transfer -c 127 -i 3000000000 -o 5400000000 -m 2 -F`.

But the output from the oscillators will be so low, and the attenuation so high throughout the RF path, that you are unlikely to be able to see any output signal.

Moji14, 2024-10-21:
> ok, from both answers - if i understand correctly...
>
> carrier wave is already there and writing any constant value in I(and 0 in Q or other way arround) would simply 'scale' carrier, meaning that [...]

Looking just at U17 and U4 datasheets it's reasonable up to 6.9GHz but you also have to account for the rest components of the Hackrf. If it was designed only up to 6 GHz, as @martinling said, that means you'll lose performance and get many undesired effects, like noise, intermodulation and so on.
If you want to go above 6 GHz with your Hackrf you might be able to use/combine with a C, X or Ku band LNA's (or even LNB's) without having to worry about the internal electronics.
It is possible to find mixers up to 325 GHz, but that is a different story :)

luoGanda, 2024-10-21:
i have an 12GHz lna, didn't test it yet though...
would not be bad having one till eg 999THz...

ok, i think this was well answered,
so - closing this thread.

closed 2024-10-21 by luoGanda