greatscottgadgets/hackrf #1478: "No HackRF boards found"

billmilosz, 2024-09-15:
### What would you like to know?

I have a recently purchased HackRF One  that I want to use.  The instructions on use are vague, but it seems that prior to connecting my Portapack H2  I need to update the firmware on the HackRF One board. 

I have a Dell notebook running Ubuntu MATE.  I have my HackRF One board connected to the Dell PC by a USB cable.  Green, yellow and red LEDs are lit on the HackRF One board.

I am following instructions from a YouTube video on how to update the firmware on my HackRF One.  The first instruction is to connect the HackRF One to a Linux computer by USB and type "hackrf_info" in a terminal window. 

So, I type **hackrf_info** in a terminal on the Dell Ubuntu computer and I get:

    hackrf_info version :2023.01.1
    libhackrf version 2023.01.1 (0.8)
    no HackRF boards found.

So how do I proceed?

straithe, 2024-09-15:
Portapack uses its own firmware. For support with that third-party add-on please contact the person you purchased it from.

As for the HackRF One board not being seen at this moment, I suggest trying a few different USB cables.

billmilosz, 2024-09-15:
Portapack is not yet installed.  I will install it after updating the firmware.  I tried six different USB cables, and all six work with other devices. I also tried all of the USB ports on the Ubuntu PC, all of which work just fine with other devices.

The Ubuntu PC needs to be able to "see" the HackRF One board before I can do anything with it.  This is a brand new board, purchased 10 days ago from HackerWarehouse.com

straithe, 2024-09-18:
You may need to try more USB cables. Please see our documentation for more information: https://hackrf.readthedocs.io/en/latest/usb_cables.html

martinling, 2024-09-18:
@straithe I think you missed that @billmilosz has already done that.

@billmilosz Can you get the device to show up in DFU mode? To do so, hold the DFU button whilst pressing and releasing the RESET button.

Then in Ubuntu if you run `lsusb` it should show up as:
`1fc9:000c NXP Semiconductors LPC4330FET180 [ARM Cortex M4 + M0] (device firmware upgrade mode)`

If it doesn't show up in DFU mode then something is faulty and you'll need to follow up with Hacker Warehouse for an RMA.

billmilosz, 2024-09-18:
I tried 6 different USB cables, all of which work well in other uses.
But I then tried ANOTHER cable - a USB "A" to USB "C" cable with a USB "C" to micro USB adapter.
This one worked, immediately.
Go figure.

straithe, 2024-09-19:
Happy to hear! I'll close this issue as completed.

closed 2024-09-19 by straithe

markotime, 2025-01-06:
hackrf board: the arm cortex device shows up in lsusb, indicating the board i in dfu mode, but any dfu command returns "no hackrf board found".  ?? thanks

straithe, 2025-01-06:
> hackrf board: the arm cortex device shows up in lsusb, indicating the board i in dfu mode, but any dfu command returns "no hackrf board found". ?? thanks

Please open a new issue for your concern.