greatscottgadgets/hackrf #1480: Brand new HackRF One out of box failure

Manishaalk, 2024-09-17:
### Have you read the HackRF [documentation](https://hackrf.readthedocs.io/en/latest/)?

no

### What outcome were you hoping for?

Bought two HackRF One. One them is not working. Signal transmitted from GNU radio to HackRF One and the HackRF One should transmitt the signal.

### What outcome actually happened?

When sent signal from GNU radio to HacRF One, no signal transmitted from HackRF One. All required LEDs including TX are on. We are using RT L-SDR for receiving signals.

### What operating systems are you seeing the problem on?

Linux

### What is the output of ```hackrf_info```?

N/A

### Are you using any third-party software?

GNU Radio

### Are you using any third-party hardware?

RTL-SDR

straithe, 2024-09-18:
Can you please add the output from hackrf_info for both of your boards to this issue?

Manishaalk, 2024-09-18:
I don't know how to get it. Can you please provide me step by step nstuctions to get the output hackrf_info? Thanks

Manishaalk, 2024-09-18:
**### Infor from Not working(defective) HackRf One**

(base) C:\Users\ManishaDesai> hackrf_info
hackrf_info version: 2024.02.1
libhackrf version: 2024.02.1 (0.9)
Found HackRF
Index: 0
Serial number: 000000000000000066a062dc276b579f
Board ID Number: 2 (HackRF One)
Firmware Version: 2024.02.1 (API:1.08)
Part ID Number: 0xa000cb3c 0x005d475c
Hardware Revision: r10
Hardware appears to have been manufactured by Great Scott Gadgets.
Hardware supported by installed firmware:
    HackRF One

**### Infor from Working HackRf One**

hackrf_info version: 2024.02.1
libhackrf version: 2024.02.1 (0.9)
Found HackRF
Index: 0
Serial number: 000000000000000066a062dc2770579f
Board ID Number: 2 (HackRF One)
Firmware Version: 2024.02.1 (API:1.08)
Part ID Number: 0xa000cb3c 0x006f435e
Hardware Revision: r10
Hardware appears to have been manufactured by Great Scott Gadgets.
Hardware supported by installed firmware:
    HackRF One

Manishaalk, 2024-09-18:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you have verified the sender's email address and know the content is safe.

### Infor from Not working(defective) HackRf One
(base) C:\Users\ManishaDesai> hackrf_info
hackrf_info version: 2024.02.1
libhackrf version: 2024.02.1 (0.9)
Found HackRF
Index: 0
Serial number: 000000000000000066a062dc276b579f
Board ID Number: 2 (HackRF One)
Firmware Version: 2024.02.1 (API:1.08)
Part ID Number: 0xa000cb3c 0x005d475c
Hardware Revision: r10
Hardware appears to have been manufactured by Great Scott Gadgets.
Hardware supported by installed firmware:
HackRF One
### Infor from Working HackRf One
hackrf_info version: 2024.02.1
libhackrf version: 2024.02.1 (0.9)
Found HackRF
Index: 0
Serial number: 000000000000000066a062dc2770579f
Board ID Number: 2 (HackRF One)
Firmware Version: 2024.02.1 (API:1.08)
Part ID Number: 0xa000cb3c 0x006f435e
Hardware Revision: r10
Hardware appears to have been manufactured by Great Scott Gadgets.
Hardware supported by installed firmware:
HackRF One

martinling, 2024-09-18:
Are you able to share a screenshot of the GNU radio flowgraph you are using?

Manishaalk, 2024-09-18:
Hi,

I bought two HackRF one units. One of them is working fine with the same flow graph. I don't think GNU radio flow graph has any issue. It is HackRF One unit.

martinling, 2024-09-18:
I can't help diagnose what the problem might be without seeing the software setup.

If you're already certain that it's a hardware fault then you should ask the reseller you purchased from about arranging a return.

straithe, 2024-09-19:
Please contact Canada Robotix again and give them a link to this issue thread. The next step is for them to contact us and we will work through the RMA process with them directly.

Manishaalk, 2024-09-19:
Thanks, I have emailed Canada Robotix to contact you and provided them the link to issue. Hoping to get no charge replacement/repair for the defective HackRF one soon.

Manishaalk, 2024-10-28:
Hi, I sent the HackRF One to Canada Robotix.  They have tested and as per them it is working fine. I received back and tested with my flow graph it is not transmitting the signal. Could you please help me. Below is the HackRF One info:

(base) C:\Users\ManishaDesai> hackrf_info
hackrf_info version: 2024.02.1
libhackrf version: 2024.02.1 (0.9)
Found HackRF
Index: 0
Serial number: 000000000000000066a062dc276b579f
Board ID Number: 2 (HackRF One)
Firmware Version: 2024.02.1 (API:1.08)
Part ID Number: 0xa000cb3c 0x005d475c
Hardware Revision: r10
Hardware appears to have been manufactured by Great Scott Gadgets.
Hardware supported by installed firmware:
    HackRF One

straithe, 2024-10-28:
As stated by Martin previously, we will need to see your flow graph to verify your request.

Manishaalk, 2024-11-08:
I have attached the information from flow graph for your reference. The signal is generated from flow graph at CF of 162MHz  and received by HackRF One. I checked the transmitted signal onnSPectrum analyzer. It is transmitting at CF of 2.7GHz.

[image]

miek, 2024-11-08:
All of the gains are set to zero so that would explain why it's not transmitting anything. As a starting point, leave the RF Gain at 0 and set both the IF Gain and BB Gain to 32.

Manishaalk, 2024-11-11:
I set IF and BB gain to 32 as advised. Still no output.
[HackRF One Test - 11Nov2024.docx](https://github.com/user-attachments/files/17706839/HackRF.One.Test.-.11Nov2024.docx)

miek, 2024-11-11:
OK, we'd still need to see the whole flowgraph to debug more.

You could also try running `hackrf_transfer -c 127 -f 162e6 -x 47` from the command-line instead, to exclude any flowgraph issues.

Manishaalk, 2024-11-11:
Here is the result after running the command:
[image]

miek, 2024-11-11:
What happened to your previous comment? It showed `hackrf_transfer` running successfully and the output on the spectrum analyzer looked correct to me. What changed after that?

Manishaalk, 2024-11-11:
That was from the working HackRF One. We have two HackRF One units. One is working and other one is not

miek, 2024-11-11:
OK, so `HackRF not found` is a different issue to before. Previously you were able to communicate with both HackRFs using `hackrf_info` and running the flowgraph, so double check your setup is still correct: make sure the device is plugged in, the cable is good, and the USB light is on.

miek, 2024-11-12:
Great, that output looks good to me - you've got the expected amount of power at around 162 MHz, so it looks like the hardware is functioning correctly.

Manishaalk, 2024-11-12:
So what could be the reason for One HackRF One working fine with our flow graph and other not?

miek, 2024-11-12:
I don't know, but unless you're willing to show us the flowgraph (or some minimal version that displays the same problem) then there's no way I can figure that out.

closed 2025-02-20 by mossmann