greatscottgadgets/hackrf #1481: Unwanted Carrier generated at central frequency in TX mode

alessiodilibertoeng, 2024-09-18:
### What type of issue is this?

permanent - occurring repeatedly

### What issue are you facing?

Hi,
when I use the HackRF One in TX mode, there is always an UNWANTED CARRIER generated at the CENTRAL FREQUENCY, even if I am not generating it (e.g. SSB modes).

The **issue is more predominant at the HF frequencies** (max is between 5 and 15 MHz) in conjunction with the +14dB onboard power amplifier enabled (the MGA81563 onboard). However this carrier gets and lower lower towards higher frequencies (essentially nothing from around 100 MHz upwards).

Here are the output technical data with **hackrf_info command**

**hackrf_info version: 2023.01.1
libhackrf version: 2023.01.1 (0.8)
Found HackRF
Index: 0
Serial number: 000000000000000017c467dc331a4cc3
Board ID Number: 2 (HackRF One)
Firmware Version: 2023.01.1 (API:1.07)
Part ID Number: 0xa000cb3c 0x006a4368
Hardware Revision: older than r6
Hardware supported by installed firmware:
    HackRF One**

This issue is very annoying and very predominant especially when I connect a power amplifier at the HackRF output and in conjunction with the +14dB onboard power amplifier enabled (the MGA81563 onboard).

This is a big issue since I want to generated a SSB modulation but, because of this issue, there is this unwanted carrier on top of the SSB and that is a big problem in terms of modulation and more in terms of unwanted output power after a linear amplifier!

Is there any way to solve it? To what is this issue related?

Thanks in advance for you support,
Alessio

### What are the steps to reproduce this?

I am using SDRangel to use HackRF One (however this issue is also present when using GNU Radio).

I enable the +14dB onboard power amplifier enabled (the MGA81563 onboard). 

I simply turn TX on (the led TX turns on as well, so confirmed it is in TX mode) without any modulation, just TX on. I have tried also generating SSB transmission without any voice to modulate the signal in order to confirm that the behaviour is modulation independent.

I increase the VGA gain and I gradually see this unwanted carrier at the central frequency.

straithe, 2024-11-12:
Please view https://hackrf.readthedocs.io/en/latest/faq.html#what-is-the-big-spike-in-the-center-of-my-received-spectrum

alessiodilibertoeng, 2024-11-12:
2 MONTHS JUST TO HAVE A COPY AND PASTE ANSWER THAT EVEN DOESN'T HELP ME AT ALL???
YOU EITHER ARE NOT ABLE TO READ AND/OR DON'T KNOW WHAT YOU ARE WRITING DOWN. I THINK BOTH. 
COMPLETELY FUTILE SUPPORT

straithe, 2024-11-12:
**Be kind.** I tried to answer the best I could from the information you gave. It sounds like you are having a problem with every center frequency and this information is just one step in diagnosing and understanding the problem. 

Now, to start again, you are saying that the spike is not what you are seeing, correct?

alessiodilibertoeng, 2024-11-13:
Let's start again...

As written in **What issue are you facing?** section and as explained in **What are the steps to reproduce this?** section, in order to reproduce such a behaviour, I am facing this issue **when I am in transmission (TX mode)**, so nothing to do with what described in the link you sent me about the usual DC spike in the middle.

I am generating a **SSB (Single Side Band) signal, so because of its nature there should not be any generated carrier with it (https://en.wikipedia.org/wiki/Single-sideband_modulation): do you agree?**

**Now, no matter the center frequency I want to generate such a SSB signal, the issue is that the carrier (that should be suppressed) is still present. This is the issue.** However it looks like the lower the frequency the worst.

**You can simply see this behaviour using HackRF One generating a SSB in TX mode connected to a Spectrum Analyser with a 50 ohm coax cable (I put in between a 3dB attenuator just to reduce possible mismatch). You will see that the carrier frequency (that again should be suppressed or at least a good starting value is more than 40dB in suppression, https://www.analog.com/en/resources/analog-dialogue/articles/single-sideband-upconversion-of-quadrature-dds-signals.html) is still there.**

**The thing that I have realized in these 2 months is that the SSB Carrier suppression value here in HackRF One is equal or worst than 30 dB because of the architecture and the components used in the design. Could you confirm this values?**

martinling, 2024-11-13:
When using any IQ modulator, there will always be some residual carrier at the LO frequency, due to inherent imperfections of the modulator itself, and the I and Q DACs.

This residual carrier will appear as a "spike" in the centre of the transmit bandwidth which is directly analogous to that seen in receive mode. The causes are the same: DC offset of the DACs, IQ amplitude imbalance, and inaccuracy of the 90 degree phase shift applied to the LO signals in the quadrature mixer.

There are two main things you can do to mitigate this effect.

One is to correct for these imperfections in software. If you have a spectrum analyzer on which to monitor the output, then you can experiment with:
- Adding/subtracting a DC offset to each of the I & Q signals to compensate for DC offset of the DACs.
- Applying a scale factor to one of the I/Q signals to compensate for I/Q amplitude imbalance.
- Applying a phase shift to one of the I/Q signals to compensate for error in the 90 degree phase shift.

The math involved is essentially the same as for receive-side correction as in the article that @Moji14 linked. By finding the right parameters for your individual setup, it should be possible to substantially reduce the residual carrier.

The other thing that you can do is to shift the remaining carrier spike well away from your intended SSB signal.

With the wide bandwidth of the HackRF, there is no need to modulate your AF signal directly into the few kHz adjacent to the LO, as would be done in an analog SSB modulator. You can instead shift your SSB signal to be as much as several MHz from the HackRF's centre frequency, in either direction.

At HF frequencies, that's enough to put the unwanted residual carrier well outside the passband of the rest of your TX path. If you're amplifying the output then you should have at least a LPF after the PA stage. If you shift your SSB signal to e.g. -5MHz and set the HackRF centre frequency to 5MHz above your target frequency, then the residual carrier will be 5MHz above and well suppressed by the LPF.

alessiodilibertoeng, 2024-11-13:
Thanks both @Moji14 and @martinling for your very detailed answers, very much appreciated.

Furthermore, @martinling very interesting approach you described in your additional improvement: 

" **With the wide bandwidth of the HackRF, there is no need to modulate your AF signal directly into the few kHz adjacent to the LO, as would be done in an analog SSB modulator. You can instead shift your SSB signal to be as much as several MHz from the HackRF's centre frequency, in either direction.

At HF frequencies, that's enough to put the unwanted residual carrier well outside the passband of the rest of your TX path. If you're amplifying the output then you should have at least a LPF after the PA stage. If you shift your SSB signal to e.g. -5MHz and set the HackRF centre frequency to 5MHz above your target frequency, then the residual carrier will be 5MHz above and well suppressed by the LPF.** ".

Nevertheless, PCB paths are not always symmetrical and same lenghts, so of course I will try to obtain the max on the digital side optimizing the algorithm.

Best regards

Moji14, 2024-11-13:
Oh at the end it seems that you found your own solution... Will you share it as good as you showed your frustration? 😉

alessiodilibertoeng, 2024-11-13:
@Moji14 Unfortunately not found a solution yet, but I will start experimenting using your and @martinling hints in order to solve this issue.

The other problem now is about when I will have again time...but that's another story :smile:

Live long and prosper :vulcan_salute:

closed 2025-02-20 by mossmann