greatscottgadgets/hackrf #1487: Cannot update firmware on my HackRF One

DerrickLau, 2024-10-01:
### Have you read the HackRF [documentation](https://hackrf.readthedocs.io/en/latest/)?

no

### What outcome were you hoping for?

That I am allowed to update the firmware on my HackRF One so that I can use it to transmit via SDRAngel 7.19.1 on Windows 10

### What outcome actually happened?

C:\PothosSDR\bin>hackrf_spiflash.exe -d 0000000000000000f75461dc287c23c3 -w C:\Derrick\AmateurRadio\firmware-bin\hackrf_one_usb.bin -v
File size 42248 bytes.
Checking target device compatibility
Unknown Board IDCompatibility test failed.

### What operating systems are you seeing the problem on?

Windows 10 v22H2 OS Build 19045.4894

### What is the output of ```hackrf_info```?

File size 42248 bytes.
Checking target device compatibility
Unknown Board IDCompatibility test failed.

C:\PothosSDR\bin>hackrf_info -?
hackrf_info version: git-7047252
libhackrf version: git-7047252 (0.6)
Found HackRF
Index: 0
Serial number: 0000000000000000f75461dc287c23c3
Board ID Number: 4 (Unknown Board ID)
Firmware Version: 2023.01.1 (API:1.07)
Part ID Number: 0xa000cb3c 0x004f4f66

### Are you using any third-party software?

Pothos SDR

### Are you using any third-party hardware?

N/A

dmaltsiniotis, 2024-10-02:
If I were to guess here, the issue is that you're using a very old version of the hackrf tools, and specifically hackrf_spiflash.exe (the version you're using appears more than 3 years old), while also trying to flash a newer board version that it doesn't know about. The Pothos folks seems to have stopped updating their SDR package some years ago unfortunately.

Good news however, you can download the latest version of the Windows tools from right here at the [latest nightly build](https://github.com/greatscottgadgets/hackrf/actions/runs/11105787227/artifacts/1995292502):

[image]

Alternatively, you can override the board id check with the --no-check flag:

`-i, --no-check: Skip check for firmware compatibility with target device.`

`hackrf_spiflash.exe -d 0000000000000000f75461dc287c23c3 -w C:\Derrick\AmateurRadio\firmware-bin\hackrf_one_usb.bin -v -i`

I would very much recommend using the latest version of the host tools though.

DerrickLau, 2024-10-02:
Downloaded the latest nightly build but when I tried running hackrf_info.exe got:
C:\Derrick\AmateurRadio\hackrf-tools-windows>hackrf_info
hackrf_info version: git-17f3943
libhackrf version: git-17f3943 (0.9)
Found HackRF
Index: 0
hackrf_open() failed: Access denied (insufficient permissions) (-1000)

I ran this using the command prompt as Administrator:
[image]

dmaltsiniotis, 2024-10-02:
Sometimes, "Access denied" doesn't mean a Windows permission issue, but rather that the USB device is busy. Try pressing the reset button on the HackRF One or unplug and re-plug the USB cable and run hackrf_info again to see it makes any difference.

I also downloaded the nightly build and ran it to make sure it wasn't some temporary nightly issue:

[image]

DerrickLau, 2024-10-02:
Thank you kindly.  I ran the flash command but my Firmware version did not change.  Where did you get your firmware from?

C:\Derrick\AmateurRadio\hackrf-tools-windows>hackrf_info
hackrf_info version: git-17f3943
libhackrf version: git-17f3943 (0.9)
Found HackRF
Index: 0
Serial number: 0000000000000000f75461dc287c23c3
Board ID Number: 4 (HackRF One)
Firmware Version: 2023.01.1 (API:1.07)
Part ID Number: 0xa000cb3c 0x004f4f66
Hardware Revision: r9
Hardware appears to have been manufactured by Great Scott Gadgets.
Hardware supported by installed firmware:
    HackRF One

C:\Derrick\AmateurRadio\hackrf-tools-windows>hackrf_spiflash.exe -w hackrf_one_usb.bin
Error opening file hackrf_one_usb.bin

C:\Derrick\AmateurRadio\hackrf-tools-windows>hackrf_spiflash.exe -w C:\Derrick\AmateurRadio\firmware-bin\hackrf_one_usb.bin
File size 42248 bytes.
Checking target device compatibility
Erasing SPI flash.
Writing 42248 bytes at 0x000000.

C:\Derrick\AmateurRadio\hackrf-tools-windows>hackrf_spiflash.exe -w C:\Derrick\AmateurRadio\firmware-bin\hackrf_jawbreaker_usb.bin
File size 34456 bytes.
Checking target device compatibility
Compatibility test failed.

C:\Derrick\AmateurRadio\hackrf-tools-windows>hackrf_spiflash.exe -w C:\Derrick\AmateurRadio\firmware-bin\hackrf_rad1o_usb.bin
File size 71700 bytes.
Checking target device compatibility
Compatibility test failed.

C:\Derrick\AmateurRadio\hackrf-tools-windows>hackrf_info
hackrf_info version: git-17f3943
libhackrf version: git-17f3943 (0.9)
Found HackRF
Index: 0
Serial number: 0000000000000000f75461dc287c23c3
Board ID Number: 4 (HackRF One)
Firmware Version: 2023.01.1 (API:1.07)
Part ID Number: 0xa000cb3c 0x004f4f66
Hardware Revision: r9
Hardware appears to have been manufactured by Great Scott Gadgets.
Hardware supported by installed firmware:
    HackRF One

miek, 2024-10-03:
After running the flash command you need to reset the HackRF before it will show the new version, by either pressing the reset button or unplugging & plugging it back in.

DerrickLau, 2024-10-03:
Thanks!

C:\Derrick\AmateurRadio\hackrf-tools-windows>hackrf_info
hackrf_info version: git-17f3943
libhackrf version: git-17f3943 (0.9)
Found HackRF
Index: 0
Serial number: 0000000000000000f75461dc287c23c3
Board ID Number: 4 (HackRF One)
Firmware Version: 2024.02.1 (API:1.08)
Part ID Number: 0xa000cb3c 0x004f4f66
Hardware Revision: r9
Hardware appears to have been manufactured by Great Scott Gadgets.
Hardware supported by installed firmware:
    HackRF One

closed 2024-10-03 by DerrickLau