greatscottgadgets/hackrf #1489: Cannot transmit on HackRF One with SDRAngel

DerrickLau, 2024-10-03:
### Have you read the HackRF [documentation](https://hackrf.readthedocs.io/en/latest/)?

yes

### What outcome were you hoping for?

To be able to transmit voice to my crappy BaoFeng UX-VR-9 walkie talkie.

### What outcome actually happened?

Nothing transmitted

### What operating systems are you seeing the problem on?

Windows 10 version 22H2 (OS build 19045.4894)

### What is the output of ```hackrf_info```?

C:\Derrick\AmateurRadio\hackrf-tools-windows>hackrf_info
hackrf_info version: git-17f3943
libhackrf version: git-17f3943 (0.9)
Found HackRF
Index: 0
Serial number: 0000000000000000f75461dc287c23c3
Board ID Number: 4 (HackRF One)
Firmware Version: 2024.02.1 (API:1.08)
Part ID Number: 0xa000cb3c 0x004f4f66
Hardware Revision: r9
Hardware appears to have been manufactured by Great Scott Gadgets.
Hardware supported by installed firmware:
    HackRF One

### Are you using any third-party software?

SDRAngel v7.19.1 
I start it.  Then add a transmission device, which I set to my HackRF One.  I set the frequency to   I then create a transmission channel,  Frequency of 435 MHz.  I go to the NFM modulator panel and press the microphone panel to on.  In Preferences->Audio, I set output device to my Realtek Microphone and Speaker, and input device to my Realtek Microphone and Speaker.  I press the play button in the device window and speak into my microphone, but my walkie talkie doesn't pick up anything.
[image]

### Are you using any third-party hardware?

No

straithe, 2024-11-12:
Please know that capturing a signal and sending that same signal requires more steps than what you've done here. Please watch https://youtu.be/pZAkhFmeLqU for more details.

miek, 2025-06-10:
I went ahead and reproduced the setup myself, but it all seems to be working for me with the latest version of SDRAngel (7.22.7). It looks like there have been some HackRF support changes in SDRAngel recently, so it may be worth trying an upgrade.

Also, I notice in your screenshot that the TX waterfall isn't showing any modulation. Unless the screenshot was taken when you weren't talking into the mic, there may have been an audio device setup issue. Here's how mine looks for comparison when I whistle into the mic:

[image]

Notice the sinusoidal wobble of the tone in the waterfall (I adjusted the settings a little to make it clearer) and also the level of the VU meter in the NFM Modulator window.

Also, it's worth setting a slight frequency offset to avoid the DC spike.

closed 2025-06-10 by miek