greatscottgadgets/hackrf #1491: Cron won't run hackrf_transfer

dudelaksdfjlkaz0102, 2024-10-10:
### Have you read the HackRF [documentation](https://hackrf.readthedocs.io/en/latest/)?

yes

### What outcome were you hoping for?

cron find hackrf_transfer & execute.

hackrf_transfer executes from command like without a problem.

### What outcome actually happened?

cron debug output shows:
/bin/sh: 1: hackrf_transfer: not found

Crontab -l shows
* * * * * hackrf_transfer -h >/home/pi/error.log 2>&1

### What operating systems are you seeing the problem on?

raspbian legacy lite

### What is the output of ```hackrf_info```?

hackrf_info version: git-cc7f5996 (0.5)

### Are you using any third-party software?

NO
Hackrf was installed with
cmake
make
sudo make install
sudo ldconfig

### Are you using any third-party hardware?

NO

pointy56, 2024-10-10:
It is likely that cron cannot find hackrf_transfer - from a quick Google search:
Ensure the script or command your cron job tries to run has the appropriate permissions and that you use absolute paths. Cron jobs run in a limited environment with a minimal PATH definition, so specifying the full path to any command or script is crucial.

closed 2024-10-10 by mossmann

dudelaksdfjlkaz0102, 2024-10-10:
> It is likely that cron cannot find hackrf_transfer - from a quick Google search: Ensure the script or command your cron job tries to run has the appropriate permissions and that you use absolute [...]

What path do I use for hackrf_transfer?

dmaltsiniotis, 2024-10-10:
> What path do I use for hackrf_transfer?

On your command line / terminal type: `which hackrf_transfer` and it will show you the full path of where the command is located. You'll want to use that full path when calling it from cron.

On a side note, it's kind of unusual to be calling hackrf_transfer from a schedule like cron, and I'm assuming the example you listed above is just for testing since -h doesn't do anything but print the help text.