greatscottgadgets/hackrf #1492: The spectrum analyzer results are different from the GNU RADIO spectrum display results.

zhiyouyan, 2024-10-17:
### What would you like to know?

I designed an AM modulated signal on GNU Radio, and the results observed on the spectrum analyzer and GNU Radio are different.
[image]

Moji14, 2024-10-17:
Can you elaborate on the problem?
Are you using hackrf hardware to transmit an AM?
Can you specify some numbers like carrier freq, modulator freq, etc?
Can you provide the spectrum analyzer output?

zhiyouyan, 2024-10-17:
[image]

zhiyouyan, 2024-10-17:
[image]

miek, 2024-10-28:
By the way, your image uploads didn't work. We'll still need more information to be able to help.

closed 2025-07-03 by miek