greatscottgadgets/hackrf #1495: How to improve modulated signal on Carrier frequency when modulated signal is 15 - 20 Mhz

smileyrekiere, 2024-10-27:
### What would you like to know?

We are trying to modulate 5 - 20 Mhz on a carrier frequency at low power (-64 dBm). 
When modulated 5 Mhz with a tx sample rate of 20 Mhz we get the expected signal measured by the signal hound .  The average power becomes less then -64 dBm  as the signal becomes "rounded".   We have also pushed the tx sample rate to 24 Mhz and did see some flatting between 13 Mhz and 15 Mhz.     The calculation of the modulated chirp uses Euler's formula.  The triangle input is the same for all modulated signals.   The phasors are not being calculated differently based on the modulated frequency. 

Basically not getting consistent power across the chirp at higher modulated frequencies.   Would like to achieve the flat signals as occurred with 5 Mhz or at least understand why ?  Even the I/Q graphs are a bit rounded on the ends which implies the maths are a bit off. 

` rf_helper::phasor *= std::exp(1i * M_PI * (rf_helper::f_mod / rf_helper::sample_rate) * rf_helper::triangle());`

[image]

As the  modulated signal becomes greater then 15 Mhz the signal becomes for "rounded"
15 Mhz
[image]

18 Mhz 
[image]

20 Mhz

[image]

miek, 2024-10-27:
This is normal. As part of the transmit signal chain, there is a filter at baseband to remove aliases. It's not realistic to make a filter that cuts off very sharply, so it has to roll off slowly and, to get enough attenuation of the aliases, it has to encroach on the edges of the signal. What you're seeing is the shape of that filter rolling off.

If you want to flatten out the transmission, you will need to alter the signal that you generate in software and reduce the amplitude in the centre of the band.

smileyrekiere, 2024-10-28:
First of all, I agree that the software generation of the signal may be the issue.  That I am failing to realize is the adjustment of the software generation of the signal.  Why at 5 Mhz is the signal flattended and at 20 Mhz is is not?

```
If you want to flatten out the transmission, you will need to alter the signal that you generate in software and reduce the amplitude in the centre of the band.
```

miek, 2024-10-28:
I'm not saying there's anything wrong with your signal generation, just that if you want a perfectly flat output you need to calibrate & correct for that in software.

The reason why the 5 MHz chirp is flat is just that it's avoiding the roll-off at the edges. If you take the 15/18/20 MHz images and imagine cropping to the centre 5 MHz, they look pretty flat there too.

Moji14, 2024-10-28:
I just saw this comment and I found it interesting.
Can you tell the modulator frequency? It always helps to know all your signals characteristics. 
If you're plotting AM power vs Time why are you taking about the filters response on frequency domain instead of time domain? 
Wouldn't be better to analyze the filter on the freq. domain?

smileyrekiere, 2024-10-28:
@miek , thank-you for clarification,  I  see that the I/Q values also have a similar roll off at the higher frequencies. 

@Moji14 
The carrier frequency is the 5.53 Ghz ,  the modulated frequency on the carrier is between 5 - 20 Mhz.   the second image is FM vs Time which shows the modulated frequency on the carrier.   The carrier is listed at the right as the center frequency.

Moji14, 2024-10-29:
> @miek , thank-you for clarification, I see that the I/Q values also have a similar roll off at the higher frequencies.
>
> @Moji14 The carrier frequency is the 5.53 Ghz , the modulated frequency on the [...]

I see, correct me If I'm wrong.
You're trying to make a chirp with Hackrf that sweeps from 5535 to 5550 MHz with a total sweep of 15 MHz with a modulating signal of about 20 KHz. 
The first thing that comes to my mind is the Nyquist criteria... In your previous comment you mentioned that "We have also pushed the tx sample rate to 24 Mhz" but the ADC (MAX5864) datasheet specifies a max sampling rate of 22Msps, and that shoud only give room to sample a baseband signal of 11MHz without aliasing. Your chirp bandwidth doesn't give room even for subsampling.

@miek Also mentioned the baseband filters... If I'm not wrong that filter is located at the transceiver (MAX2837) and I think it can be increased the roll off frequency or even bypassed (not sure if hackrf_tranfer.c allows you to do that or the Hackrf firmware is responsible for that). That could be an option to see if you can improve your RF output.

If you need that flatness on your RF output I think it'll be a good idea to split your wide chirp into 2 small ones making 1 swep say for example from 0 to 7.5 MHz  on baseband and then up-convert to RF twice.

smileyrekiere, 2024-10-29:
@miek , adjusting the I/Q values (using crude math at this point) to adjust for the filter cutoffs at the higher frequencies was able to achieve more of a square wave.  

I had kept trying to adjust the beginning and ending of the chirp not realizing it was not possible until you pointed out the filter was playing a role so needed to adjust (reduce) the power of the middle of the chirp.  I didn't make the connection then you pointed out that from -2.5 Mhz to 2.5 Mhz the chirps were fine regardless of the chirp size.  

[image]

smileyrekiere, 2024-10-29:
for comparison here is the signals without the I/Q adjustments (so this would be the pre slide and the above would be post)
[image]

smileyrekiere, 2024-10-29:
closing out ,  thank-you for the help.

closed 2024-10-29 by smileyrekiere