greatscottgadgets/hackrf #1501: Issues updating HackRF One R1 board from 2014 to newer firmware

TK5005, 2024-11-08:
### What type of issue is this?

permanent - occurring repeatedly

### What issue are you facing?

I am trying to update my HackRF One from 2014 to newer firmware. I am able to update firmware to RAM via dfu-util with no problem and then use hackrf_cpld to program the bitstream. When I try to update the firmware utilizing hackrf_spiflash I get no errors, however when I reboot the board the older firmware persists. Trying to use hackrf_cpld in normal mode gives me a pipe error.

### What are the steps to reproduce this?

hackrf_spiflash -w hackrf_one_usb.bin
hackrf_cpldjtag -x firmware/cpld/sgpio_if/default.xsvf

Restart radio

miek, 2024-11-08:
You don't need to update the CPLD as it's included in the firmware now, so don't worry about that.

If running `hackrf_spiflash` and then rebooting the device doesn't stick, then instead you could try running `hackrf_spiflash -c -w hackrf_one_usb.bin`. This clears any write protect bits that can prevent the update succeeding.

TK5005, 2024-11-08:
Worked perfectly! I appreciate the assistance!

closed 2024-11-08 by TK5005