greatscottgadgets/hackrf #1504: Osmocom use

ikibalamRF, 2024-11-14:
### Have you read the HackRF [documentation](https://hackrf.readthedocs.io/en/latest/)?

no

### What outcome were you hoping for?

transmit and receive a cosine using hackrf one  and using the block osmocom

### What outcome actually happened?

In the receiver I try to demodulate but I don't know how the osmocom block works. At the receiver I am multiplying by a cosine to demodulate and using a low pass filter to recover the original signal which is a cosine but the signal appears to still be modulated

### What operating systems are you seeing the problem on?

windows 11

### What is the output of ```hackrf_info```?

*

### Are you using any third-party software?

GNU RADIO

### Are you using any third-party hardware?

antenna

miek, 2024-12-20:
Multiplying by a cosine doesn't demodulate a signal, it just shifts the frequency.
Can you explain a bit more about what sort of signal you're working with and what you want to do with it?

Moji14, 2024-12-25:
@ikibalamRF Judging from your comment, it seems that you're trying to demodulate an AM signal.
Yes, by multiplying and filtering is possible to demodulate a receiving AM signal. But that will only happen successfully if your LO is matched with the carrier frequency of the transmitted signal, and I think that's were your problem resides.
Check this website out https://en.m.wikipedia.org/wiki/Product_detector

ikibalamRF, 2025-01-17:
And how can I synchronize the carrier signal frequency with the local oscillator in GNU Radio?

Moji14, 2025-01-17:
@ikibalamRF 
This might help you ...
https://wiki.gnuradio.org/index.php/AM_Demod

closed 2025-04-23 by miek