greatscottgadgets/hackrf #1508: Burnt board

docjones82, 2024-11-24:
### What would you like to know?

I have a chip tho knits regulator for sub but not sure wondering if I can find out what is purple circled chip looks burned and usb and ports pack act up plugged in together 
[image]

martinling, 2024-11-27:
The part circled in purple is Q5 which is a PMOS transistor. Its purpose is to cut off the supply path from VBUS to VIN, to stop the HackRF being powered from USB when an accessory board (like PortaPack) is supplying VIN instead. It is controlled by the VBUSCTRL signal which comes from the PortaPack.

The Q5 part we currently use is Yangzhou Yangjie part [YJL2301C](https://www.21yangjie.com/pdf/mosfet/YJL2301C.pdf) which is rated for 3.4A drain current and 20V drain-source voltage. If you want to replace it, you can use any other SOT-23 PMOS transistor with the same pinout and similar ratings, there are many possible substitutes.

In normal use it should not be possible to blow Q5. There might be a fault on the PortaPack that has caused it to draw more than the rated current through Q5 while trying to charge from USB. The fact that everything still works from battery power when USB is not connected, suggests that the rest of the HackRF board is probably OK. If you replace Q5 then it should hopefully be usable again. If that works it will also confirm that there is a fault on the PortaPack side.

docjones82, 2024-11-27:
I think it was on when it got plugged together or taken apart  is what happened

closed 2025-02-20 by mossmann