greatscottgadgets/hackrf #1509: Hackrf only detected in DFU

Dragulove, 2024-11-25:
### Have you read the HackRF [documentation](https://hackrf.readthedocs.io/en/latest/)?

no

### What outcome were you hoping for?

HackRF being detected

### What outcome actually happened?

HackRF was only detected as LPC when in DFU

### What operating systems are you seeing the problem on?

Windows 11/Ubuntu

### What is the output of ```hackrf_info```?

hackrf_info version: 2024.02.1
libhackrf version: 2024.02.1 (0.9)
No HackRF boards found.

### Are you using any third-party software?

No.

### Are you using any third-party hardware?

No.

martinling, 2024-11-27:
If the HackRF is only being detected as an LPC device in DFU mode, then you can follow the instructions to [reinstall the firmware](https://hackrf.readthedocs.io/en/latest/updating_firmware.html#only-if-necessary-recovering-the-spi-flash-firmware) over DFU.

In short, you will need the `dfu-util` tool, the HackRF command line tools including `hackrf_spiflash`, and the `hackrf_one_usb.dfu` and `hackrf_one_usb.bin` files from the latest firmware release.

To upload the initial firmware to the LPC device:

`dfu-util --device 1fc9:000c --alt 0 --download hackrf_one_usb.dfu`

The HackRF should then appear in `hackrf_info` with the serial number `RunningFromRAM`.

To finish the process, write new firmware to SPI flash:

`hackrf_spiflash -w hackrf_one_usb.bin`

Dragulove, 2024-12-01:
This worked, thank you!

Dragulove, 2024-12-01:
This worked for me, Thanks!

closed 2024-12-01 by Dragulove