greatscottgadgets/hackrf #1522: Clock input voltage (minimal value)

Derso1, 2025-01-08:
### What would you like to know?

I would like to know what is the minimal range of voltage that Hackrf one can understand as a valid clock?

Today I have a signal with the range between 0 and 500 mV. Is it enough?

TKS

dmaltsiniotis, 2025-01-08:
Hi @Derso1,

500 mV may not be enough to trigger the clock. From the [official documentation](https://hackrf.readthedocs.io/en/latest/external_clock_interface.html) they state:

> The CLKIN port on HackRF One is a high impedance input that expects 3.3 V square wave at 10 MHz. Do not exceed 3.3 V or drop below 0 V on this input

Given that, if we assume the midpoint of 0-3.3 (plus some buffer) to trigger the rising edge, I would guess at least 2v to 2.5v would be needed. The data sheet for the Si5351 indicates a maximum input voltage range of -0.5 to 3.8, but it doesn't really specify a minimum that I can see.

mossmann, 2025-01-08:
The minimum input voltage swing for Si5351C is unspecified, but one HackRF One revision (r9) used 74AUP2G58GUX as an input buffer. For compatibility with all revisions, refer to the 74AUP2G58GUX documentation for minimum clock input signal levels.

Moji14, 2025-01-11:
@Derso1 The minimum voltages of the Si5351 A/B/C for a given VDD is:
(For a 0 level) Vlow = [-1, 0.99] V
For a 1 level) Vhigh = [2.31, 3.6] V
So, 500mV it's too small so far.