greatscottgadgets/hackrf #1529: No DFU capable USB device available

TUSHAR-790, 2025-02-28:
### What outcome were you hoping for?

I was trying to use it with gr-gsm but it's was not detecting the sdr so I thought I that I should update the the firmware on the hackrf one so it might solve the issue of 

### What outcome actually happened?

Trying to update firmware on my hack rf one but it keeps showing no dfu capable USB device available.
Please help me with this issue 

### What operating systems are you seeing the problem on?

LINUX BASED UBANTU

### What is the output of ```hackrf_info```?

hackrf_info version: 2023.01.1

libhackrf version: 2023.01.1 (0.8)

No HackRF boards found.

### Are you using any third-party software?

N/A

### Are you using any third-party hardware?

N/A

TUSHAR-790, 2025-02-28:
[image]

devanha, 2025-03-01:
I have a similar problem. What lights are showing on your board? I have power green, then amber and red and no other led illuminated.