greatscottgadgets/hackrf #1530: Transmitting on several frequencies not as programmed

ainabea, 2025-02-28:
### What type of issue is this?

permanent - occurring repeatedly

### What issue are you facing?

I am trying to produce noise in certain frequencies and was able to successfully produce this for some and not all the frequencies I wanted despite having the same settings for everything. I have attached the generated python code from my gnu radio file. My test set up includes using a spectrum analyzer connected via an SMA cable to the hack RF antenna port. The HackRF one is connected to my windows laptop via a USB chord.

The frequencies I am having trouble with are 880 MHz, 744 MHz, 2.12 GHz, 2.165 GHz, 627 MHz and 642 MHz. For each of these frequencies I'm expecting a bandwidth of about 30 MHz (I measured the bandwidth using the bottom of the signal curve on both sides). I am also expecting power to be between -30 dBm and -3dBm. What I am seeing is either no discernable peak in the spectrum at my expected frequencies (specifically 2.12 GHz and 2.165 GHz), or a significantly smaller peak at the other frequencies. For example, the bandwidth for the peaks I can see are about 2 MHz and have a power maximum of around -55 dBm.

Is this expected behavior for these frequencies? Or is there something wrong with my code? I am able to get my expected results in other frequencies ranging from 744 MHz and 2.185 GHz. (The 744 MHz one worked once before and did not again). 

### What are the steps to reproduce this?

To reproduce this issue I would run the code I attached and connect the antenna output on the Hack RF to a spectrum analyzer. I set my sweep value to about 9 ms. And I used the frequency and marker capabilities to zoom in on a potential peak for analysis. 

[bug_sample.txt](https://github.com/user-attachments/files/19033494/bug_sample.txt)

mossmann, 2025-03-20:
There appear to be some bugs in your flowgraph. The sample rate of 32 ksps is unsupported. HackRF One supports sample rates of 2 Msps to 20 Msps. Do not use a throttle block in a flowgraph that connects to a HackRF (or any hardware source or sink).

ainabea, 2025-03-25:
Hi @mossmann, thank you for your response. Fixing the sample rate and removing the throttle blocks fixed the issue. Now I see the noise at the appropriate frequencies. However, it flickers a lot. Is that because the sample rate is lower than the noise frequency? Also would you be able to forward me to resources that talk about how sample rate works with HackRF and descriptions for using gnuradio for transmitting with HackRF?

mossmann, 2025-06-19:
I'm glad to hear you were able to fix your issue!

For general background on how GNU Radio and HackRF work, I suggest our video series: https://greatscottgadgets.com/sdr/