greatscottgadgets/hackrf #1531: RF LED Red & "No HackRF words found."

devanha, 2025-03-01:
### What outcome were you hoping for?

I expected the device to be detected so I could flash the latest firmware.

This is an old HackRF One board purchased in 2019 and not used for years. I was hoping to use it with the Portapack H4M. I have no idea what version of the firmware is currently on the board. I've tried accessing it on two different computers (Apple Silicon and Apple Intel) and several different cables and adapters. 

### What outcome actually happened?

The board lights up with:

3v3 - Green
1v8  - Amber
RF - Red
USB - off
RX - off
TX - off

### What operating systems are you seeing the problem on?

MacOS 15.3.1 (24D70)
Kali Linux 6.11.2-amd64 

### What is the output of ```hackrf_info```?

hackers_info version 2024.02.1
libhackrf version 2024.02.1 (0.9)
No HackRF boards found.

### Are you using any third-party software?

N/A

### Are you using any third-party hardware?

N/A

aowd, 2025-03-03:
you should switch the portapack mode to the hackrf mode by press the screen's hackrf mode icon after connect the portapack h4m,then you could find hackrf.

devanha, 2025-03-03:
I have managed to flash the latest firmware after falling another posters advice and used a usb-a to usb-c cable with a usb-c to micro-usb adapter. However, after further investigation I've discovered that my board is a pre R6 version so not supported by the Portapack.  Note to self: - RTFM!

mossmann, 2025-03-20:
Is PortaPack H4M unsupported on HackRF One older than r6? If so, can you point us to the documentation you found?

closed 2025-07-03 by mossmann