greatscottgadgets/hackrf #1533: DSP hangs with remote control

good-point-lab, 2025-03-15:
### What type of issue is this?

permanent - occurring repeatedly

### What issue are you facing?

Environment: Bookworm Raspberry Pi latest OS, Raspberry Pi 3B, peer-to-peer Ethernet connection to ESP32 MCU with W5500 module. Latest Gqrx build from source.

It works fine without the remote control. It also works with the remote control, but the FFT and Waterfall freeze after 10-20 minutes. The DSP cannot be restarted from the UI. The rest of the UI remains functional, so I can restart Gqrx. This issue occurs every time.

### What are the steps to reproduce this?

Start up Gqrx with the remote control properly configured.  
Connect to the remote control server and remotely start the DSP.
Remotely change the frequency and confirm it works OK.
Leave the system for some time and observe the FFT and waterfall freeze.

martinling, 2025-03-16:
It sounds like this issue belongs on the [gqrx](https://github.com/gqrx-sdr/gqrx) repository?

good-point-lab, 2025-03-16:
Yes, it is the wrong place for the issue. Please disregard.

closed 2025-03-16 by martinling