greatscottgadgets/hackrf #1534: HackRF not detected in SDRSharp AIRSPY SDR

Janxyxy, 2025-03-18:
### What outcome were you hoping for?

I expected SDRSharp (dotnet9 build) to detect and access my HackRF One device without any issues after confirming that the device works correctly via hackrf_info and hackrf_transfer. I expected to select HackRF as input device in SDR# and start receiving data.

### What outcome actually happened?

When launching SDRSharp.dotnet9.exe, the software displays an error message:
`Cannot open HackRF device. Is the device locked somewhere?`

[image]

### What operating systems are you seeing the problem on?

Windows 11
MSYS2 toolchain used for hackrf-tools testing

### What is the output of ```hackrf_info```?

hackrf_info version: 2024.02.1  
libhackrf version: 2024.02.1 (0.9)  
Found HackRF  
Index: 0  
Serial number: 000000000000000078d063dc27806a67  
Board ID Number: 2 (HackRF One)  
Firmware Version: v2.1.0 (API:1.08)  
Part ID Number: 0xa000cb3c 0x006a4f63  
Hardware Revision: r10  
Hardware does not appear to have been manufactured by Great Scott Gadgets.  
Hardware supported by installed firmware:  
    HackRF One

### Are you using any third-party software?

SDRSharp (dotnet9 version) – official build from Airspy
MSYS2 hackrf-tools (build: 2024.02.1)
Airspy SDR drivers included with SDR#

### Are you using any third-party hardware?

HackRF One
PortaPack H4M with Mayhem firmware (USB mode enabled – confirmed working via hackrf CLI tools)

martinling, 2025-03-20:
Given that it works with `hackrf_transfer`, there doesn't seem to be anything wrong with the HackRF side of things, so this seems like an issue you'll need to raise with SDRSharp.

You could also try some of the other tools such as SDR++.

Janxyxy, 2025-03-20:
Thanks for the message! Since hackrf_transfer works, the problem is probably with SDRSharp. I’ll check it out and try SDR++.

closed 2025-03-20 by Janxyxy