greatscottgadgets/hackrf #1535: There was a problem with the hackrf_transfer command

ggggggch, 2025-03-19:
### What would you like to know?

I use the following line of command to expect the waveform to be recorded

`hackrf_transfer -r hackrf_capture.iq -f 2414000000 -s 20000000 -g 30 -n 100000000`

But the recorded waveform is as follows, which I think is problematic.

[image]

The waveform I expected is below

[image]

ggggggch, 2025-03-19:
Please help me. It's a urgent matter.

martinling, 2025-03-19:
HackRF outputs signed 8-bit samples. You need to save the file with a `.cs8` extension for inspectrum to know the format. This is covered in the [inspectrum README](https://github.com/miek/inspectrum?tab=readme-ov-file#input).

ggggggch, 2025-03-19:
来函收悉（此为系统自动恢复）The message had been received!(This is an auto-generated message.)