greatscottgadgets/hackrf #1537: error: hackrf_set_sample_rate() failed: Input/Output Error (-1000)

radiohead2024, 2025-03-21:
### What outcome were you hoping for?

save iq data to a file. 

### What outcome actually happened?

I am trying to collect iq data from a FM station (88.9 MHz) with a sample rate of 20 MHz and I keep getting an error message below. Thanks.

hackrf_transfer -r out.iq -f 88900000 -g 50 -l 32 -s 20000000
call hackrf_set_sample_rate(20000000 Hz/20.000 MHz)
hackrf_set_sample_rate() failed: Input/Output Error (-1000)

### What operating systems are you seeing the problem on?

Windows 10

### What is the output of ```hackrf_info```?

call hackrf_set_sample_rate(20000000 Hz/20.000 MHz)
hackrf_set_sample_rate() failed: Input/Output Error (-1000)

### Are you using any third-party software?

N/A

### Are you using any third-party hardware?

N/A

martinling, 2025-05-02:
It's hard to tell what the cause might have been here without more info.

- The `hackrf_info` output would have been helpful (both to see the software/firmware versions, and to see if other requests to the device work).
- Is this running directly on Windows 10, or through WSL?
- What build of the HackRF tools are you using?
- Have you by any chance been messing around with Zadig?

radiohead2024, 2025-05-06:
Here is the output for hackrf_info:
hackrf_info version: 2024.02.1
libhackrf version: 2024.02.1 (0.9)
Found HackRF
Index: 0
Serial number: 000000000000000066a062dc27a3569f
Board ID Number: 2 (HackRF One)
Firmware Version: 2024.02.1 (API:1.08)
Part ID Number: 0xa000cb3c 0x0054474a
Hardware Revision: r10
Hardware appears to have been manufactured by Great Scott Gadgets.
Hardware supported by installed firmware:
    HackRF One
hackrf_close() failed: Input/Output Error (-1000)

I am using Windows 10 and used "conda create -n radioconda -c conda-forge -c ryanvolz --only-deps radioconda" for the install.
Lastly, I used Zadig to update my USB driver.
Thanks.