greatscottgadgets/hackrf #1539: Seeking Guidance on Efficient Harmonic Sweeping and Accurate Power Measurements with HackRF One

pablogcardarelli, 2025-03-25:
### What would you like to know?

### Have you read the HackRF documentation?
Yes

### What outcome were you hoping for?
I was hoping to perform fast frequency sweeps using the HackRF One, but instead of sweeping across the entire frequency range (like from 1 MHz to 6 GHz as `hackrf_sweep `does), I wanted to sweep only specific harmonics of interest. The goal is to efficiently obtain power measurements at precisely chosen harmonic frequencies without wasting time and computational resources on irrelevant frequencies.

Additionally, I expected the power measurements at these harmonics to be accurate and consistent, with the ability to correctly interpret the output power values (in dB) as absolute power levels (like dBm). I wanted to avoid DC offset and frequency leakage issues when performing FFT on the collected I/Q data.

### What outcome actually happened?

1. The current approach of sweeping large frequency ranges (like `hackrf_sweep` does) feels a bit inefficient for my application, as I am only interested in specific harmonic frequencies.

2. Trying to directly tune to desired harmonics using `hackrf_set_freq()` is a bit cumbersome and slow, especially when I need fast switching between harmonics.

3. The power measurements at specific harmonic frequencies often seem inconsistent, even when I inject a known signal. I'm not entirely sure whether the HackRF is correctly tuned to the specified frequency during fast sweeps.

4. I’m not sure how to efficiently and accurately sweep only the frequencies I care about while maintaining consistent power readings, especially when dealing with harmonics that are spread across a wide range.

5. The dB power values reported by HackRF don’t seem to correspond to actual power levels (like dBm), and I’m not entirely sure how to interpret them or whether some form of calibration is needed.

### What operating systems are you seeing the problem on?
Ubuntu 22.04.1 LTS

### What is the output of hackrf_info?
```
hackrf_info version: 2023.01.1
libhackrf version: 2023.01.1 (0.8)
Found HackRF
Index: 0
Serial number: 0000000000000000f75461dc2a37b6c3
Board ID Number: 4 (HackRF One)
Firmware Version: 2023.01.1 (API:1.07)
Part ID Number: 0xa000cb3c 0x004f4f5f
Hardware Revision: r9
Hardware appears to have been manufactured by Great Scott Gadgets.
Hardware supported by installed firmware: HackRF One
```

### Are you using any third-party software?
No

### Are you using any third-party hardware?
Yes, I am using laboratory wave generators to produce calibrated input signals in order to validate the power measurements at specific harmonic frequencies. These signals are well-characterized and have a stable output, allowing me to test the accuracy and consistency of the HackRF’s power readings.

### Additional context or questions:

1. Is there a way to directly and efficiently tune the HackRF to specific harmonic frequencies without sweeping the entire range?

2. Can I achieve fast frequency hopping between specific harmonics without significant delay or instability?

3. How do I ensure that the HackRF is correctly tuned to the intended frequency, especially when rapidly switching between harmonic frequencies?

4. How should I interpret the power values reported by HackRF (in dB) — are they absolute values (like dBm), or do they need to be calibrated?

5. Are there best practices to avoid DC offset and frequency leakage when performing FFT on the I/Q samples, especially in fast-sweeping scenarios?

6. Does the `hackrf_sweep `utility or its underlying implementation provide a way to configure sweeping at specific harmonics only, rather than sweeping an entire band?

Any guidance on how to optimize fast, harmonic-specific sweeping and obtain accurate power measurements would be greatly appreciated!

miek, 2025-04-23:
>     1. Is there a way to directly and efficiently tune the HackRF to specific harmonic frequencies without sweeping the entire range?
>     2. Can I achieve fast frequency hopping between specific harmonics without significant delay or instability?

No, without some software (and possibly firmware) changes, it can only be configured to sweep quickly over one continuous range. Though, see below for more on a custom software option.

>     3. How do I ensure that the HackRF is correctly tuned to the intended frequency, especially when rapidly switching between harmonic frequencies?

In terms of figuring out when the tuning operation is done, there isn't a great way of doing that, you just have to have a delay before starting to use the new samples. The way `hackrf_sweep` implements that is that, because it has to send 16KB worth of samples for each frequency anyway, it just uses the very end of that chunk to run an FFT over.

One other thing to note, is that it's not always possible to find a perfect PLL configuration to tune to the desired frequency, so there will often be some small error. There's a wishlist item for reporting back the exact tuned frequency: https://github.com/greatscottgadgets/hackrf/issues/524, but if that's an issue you can also use `hackrf_set_freq_explicit` with your own tuning calculations to get an exact value.

Also, if you're not already aware, you'll want to use offset-tuning rather than tuning directly to the harmonic you want to measure, to avoid the [DC spike](https://hackrf.readthedocs.io/en/latest/troubleshooting.html#there-is-a-big-spike-in-the-center-of-the-received-spectrum).

>     4. How should I interpret the power values reported by HackRF (in dB) — are they absolute values (like dBm), or do they need to be calibrated?

The dB values are relative to the full-scale of the ADC, so you would have to do some calibration to measure actual power.

>     5. Are there best practices to avoid DC offset and frequency leakage when performing FFT on the I/Q samples, especially in fast-sweeping scenarios?

The way `hackrf_sweep` handles that is to do "interleaving" - it overlaps each tuning step so that it can avoid capturing parts around the DC spike and the filter roll-off at the edges (allowing for a tighter baseband filter for reduced aliasing).

For a 20 MHz-wide FFT, it will discard 2.5 MHz either side of DC and discard 2.5 MHz on either outer edge to avoid the DC spike and filter roll-off. It will then tune up 5 MHz and do the same, which captures the missing area around DC and the upper roll-off from the first tune, and also the lower roll-off from the next tune.

Here's a sketch of the process:

[image]

Though if all you're looking for is to measure a single harmonic per tune, then just avoid it being near the DC spike or the edge of the range.

>     6. Does the `hackrf_sweep `utility or its underlying implementation provide a way to configure sweeping at specific harmonics only, rather than sweeping an entire band?

On reviewing the code, I think it should be possible to use the underlying implementation of sweep to do what you want. `hackrf_init_sweep` takes a list of frequencies to tune to, and will loop over that list returning a block of samples for each frequency. You'll probably want to turn off the interleaving mode as it's expecting to work with a continuous range. The sweep mode inserts a small header into the sample stream to tell you which frequency it's currently tuned to, you can check the `hackrf_sweep` implementation to see how to read that.

pablogcardarelli, 2025-04-24:
Wow! Thank you so much for the incredibly detailed and helpful answer!

I really appreciate the explanation and the visual sketch, it clarified what was going on under the hood with `hackrf_sweep`. I hadn’t realized I could pass a custom list of frequencies to `hackrf_init_sweep()` like that, and it’s great to know I can disable interleaving if I'm targeting single harmonics.

I’ll definitely explore using offset tuning to avoid the DC spike and try to calibrate properly using my signal generator setup. The advice about discarding the initial samples after tuning is also something I’ll put into practice, it makes a lot of sense now that you mention how the sweep chunks are handled.

Thanks again for your help! This really helped me better understand the workflow and capabilities of HackRF.

closed 2025-04-24 by pablogcardarelli