greatscottgadgets/hackrf #1545: nooelect tiny tcxo installation

tapiotim, 2025-04-11:
### What would you like to know?

I just got the tiny tcxo for my many years old HackRF. Absolutely no documentation so where does it go? P22 at the end nearest the CLK IN? And it appears I need to do some configuration in software as well? Sorry, I'm old and the brain doesn't work as well any longer. I did search and I did not find anything specific referring to this? And which software package? I need someone to point me in the correct direction please.,

miek, 2025-04-13:
> P22 at the end nearest the CLK IN?

Yep, the diagram on the back of it should line up with the outline of the header, like this:

[image]

>And it appears I need to do some configuration in software as well?

Nope, no configuration needed, it'll be detected automatically (though you can double check it's been detected by running `hackrf_clock -i` from the command-line - more general info here: https://hackrf.readthedocs.io/en/latest/external_clock_interface.html).

tapiotim, 2025-04-13:
Thank you! That position made the most sense but, I try to not assume too
much. And I don't like letting the magic smoke out of things any longer.

Tim

closed 2025-04-23 by miek